nativescript-community/ble #124: Feature Request: Descriptor Support

drewcovi, 2019-01-16:
Perhaps a question for @farfromrefug, but I'm going to take a stab at building out support for Descriptors, largely by comparing it to cordova-ble (https://github.com/evothings/cordova-ble) where I had a fair degree of success in the past.

Any guidance, thoughts, or feedback is welcome. (Or if it's super quick and I shouldn't bother) let me know.

farfromrefug, 2019-01-17:
@drewcovi shouldn't they be already working? I think they are. What do you intend to do that does not work?

drewcovi, 2019-01-17:
@farfromrefug they appear to be commented out in various places for iOS with notes that they will be added later. I've uncommented the two lines that seem to support descriptors, and I'm looking to see if it's that simple.

https://github.com/EddyVerbruggen/nativescript-bluetooth/blob/e3a7ee2b556ce5154d32070fb787f9ddfa39b29a/src/bluetooth.ios.ts#L89

https://github.com/EddyVerbruggen/nativescript-bluetooth/blob/e3a7ee2b556ce5154d32070fb787f9ddfa39b29a/src/bluetooth.ios.ts#L107

drewcovi, 2019-02-04:
@farfromrefug just a quick update here, I'm a little tentative to jump in as there's quite a lot of churn right now (and rightfully so! great work). 

If this is a no-brainer for you, please don't let me hold you up. 

But Descriptor support would be amazing, as it really informs Characteristics (by providing units, format, and human readable description).

Thanks again for the amazing work this last month.

farfromrefug, 2019-02-05:
@drewcovi the support is almost there. Need to write the ```discoverDescriptors``` first.
If you look ```discoverServices``` and ```discoverChararcteristics``` is already there. 
If you want to add it, glad to see a PR!

Don't really have time to work on this right now. Lot of other things to do ;) But i want to!

closed 2020-12-10 by farfromrefug