nativescript-community/ble #125: Not able to detect mobile or laptop bluetooth

Suralal001, 2019-01-16:
Attaching sample code base.The demo app is not able to detect any of the bluetooth devices .i have tried with one plus 6t and one plus 1.both were not gettting detected.
[nativescript-demo.zip](https://github.com/EddyVerbruggen/nativescript-bluetooth/files/2766056/nativescript-demo.zip)

farfromrefug, 2019-01-17:
@Suralal001 Are you trying to detect BLE devices? This module does not support Bluetooth 2.0 devices. Usually when you try to detect a "phone" that s a 2.0. device

Suralal001, 2019-01-17:
I want to detect Bluetooth 2.0 3.0 and 4.0 version.is there any possible way to detect these versions.

farfromrefug, 2019-01-17:
I dont think it s possible with that module right now. @EddyVerbruggen ?

EddyVerbruggen, 2019-01-17:
Indeed, it's BLE only.

Suralal001, 2019-01-17:
but its only detecting smart watches and not the mobile devices.

farfromrefug, 2019-01-17:
@Suralal001  4.0 IS BLE. Pi works just fine with this module. Smart watch are BLE too. so this module works as expected. I think we can close that discussion

closed 2019-01-17 by EddyVerbruggen

Suralal001, 2019-02-12:
the issue i could find was my location was not turned on.now fixed