nativescript-community/ble #128: Error in connect+disconnect+connect sequence

raffaeler, 2019-02-06:
I can successfully connect and disconnect from a device.
If I try to reconnect to the same device, I get this error:
```
JS: connecting  A8:15:44:3E:1F:22  ... (busy=true)
JS: ERROR Error: Uncaught (in promise): TypeError: Cannot convert undefined or null to object
JS: TypeError: Cannot convert undefined or null to object
JS:     at Function.assign (<anonymous>)
JS:     at Bluetooth.connect (file:///data/data/org.nativescript.bluehello/files/app/tns_modules/nativescript-bluetooth/bluetooth.js:1217:20)
JS:     at Bluetooth.descriptor.value (file:///data/data/org.nativescript.bluehello/files/app/tns_modules/nativescript-bluetooth/bluetooth.common.js:74:31)
JS:     at file:///data/data/org.nativescript.bluehello/files/app/tns_modules/nativescript-bluetooth/bluetooth.common.js:52:35
JS:     at ZoneDelegate.invoke (file:///data/data/org.nativescript.bluehello/files/app/tns_modules/nativescript-angular/zone-js/dist/zone-nativescript.js:388:26)
JS:     at Object.onInvoke (file:///data/data/org.nativescript.bluehello/files/app/tns_modules/@angular/core/bundles/core.umd.js:16201:37)
JS:     at ZoneDelegate.invoke (file:///data/data/org.nativescript.bluehello/files/app/tns_modules/nativescript-angular/zone-js/dist/zone-nat...
JS: ERROR Error: Uncaught (in promise): [object Undefined]
````

My code to connect/disconnect is the following. the `btdevice` type simply holds some data that I got from the scan/discovery.
```
    public connect(device : btdevice) : Promise<void> {
        if(this._isBusy || device.isConnected) return;
        this._isBusy = true;
        console.log("connecting ", device.uuid, " ... (busy=true)");

        try {
            return bluetooth.connect({
                UUID: device.uuid,
                onConnected: data => {
                    device.isConnected = true;
                    this._isBusy = false;
                    console.log("connected: ", data, " (busy=false)");
                },
                onDisconnected: (peripheral : Peripheral) => {
                    device.isConnected = false;
                    this._isBusy = false;
                    console.log("disconnected: ", peripheral, " (busy=false)");
                }
            });
        }
        catch(e){
            this._isBusy = false;
            console.log("connect failed on ", device.uuid, " (busy=false)");
        }

        return;
    }

    public disconnect(device: btdevice) : Promise<void> {
        if(this._isBusy || !device.isConnected) return;
        this._isBusy = true;
        console.log("disconnecting ... (busy=true)");

        try{
            return bluetooth.disconnect({
                UUID: device.uuid
            }).then(() => {
                this._isBusy = false;
                console.log("(busy=false)");
                console.log("disconnected ", device.name);
            });
        }
        catch(e) {
            console.log("disconnect> error ", e);
            this._isBusy = false;
            console.log("(busy=false)");
        }
    }
```

Am I missing something? Thanks!

closed 2020-08-27 by farfromrefug

raffaeler, 2020-08-27:
no suggestions?

farfromrefug, 2020-08-27:
@raffaeler we dont even know what version you are using.
From now oni will only support `nativescript-bluetooth` beta 3.x and `@nativescript-community/ble`

raffaeler, 2020-08-27:
At that time it was the latest, then I dropped the library because of that problem. I will see if the new version is worth the try