nativescript-community/ble #129: Send pairing key (PIN) to Bluetooth device (glucose monitor)

JeffersonLPLima, 2019-02-07:
I am trying to pair my glucose monitor, but it is requesting a pairing key (6 digit PIN).  I would like to pair so that next time the monitor already recognizes my smartphone.
  **So, how can I send the PIN through the `bluetooth.connect` method?**
I can already read infromations from monitor even without entering the PIN, but the monitor does not pair.

farfromrefug, 2019-02-07:
This is out of the scope of that module. Please read the specs of your device. Everything should be written on how to connect to it. Not even sure this is a BLE device.

Tyler-V, 2019-07-19:
@farfromrefug some devices require pairing where you have to enter a PIN and will not let you interact with the device services until you have done so.

Manually pairing and entering the pin then allows you to connect to the device via UUID.

Many android apps allow device pairing within the application where you enter the PIN and are able to pair without leaving the app.

This seems integral to a bluetooth library and in scope to this module. Can we reevaluate this issue?

farfromrefug, 2019-07-19:
@Tyler-V  seems to me like you are talking about bluetooth 2.0 devices. this plugin is for ble  devices .

Tyler-V, 2019-07-19:
@farfromrefug bluetooth [LE 4.0 and 4.1 use Secure Simple Pairing (SSP)](https://security.stackexchange.com/questions/116027/difference-between-secure-simple-pairing-and-secure-connections-in-bluetooth) which is different from legacy pairing. It uses Passkey Entry, which requires you to enter a 6-digit numeric code to prevent MITM attacks.

The device I am using is a Contour Next One bluetooth glucose meter which follows the 4.0 specification.

farfromrefug, 2019-07-20:
@Tyler-V did not know about SSP on BLE! never used a device which used it.
Maybe you could make a PR?

closed 2020-08-27 by farfromrefug