nativescript-community/ble #130: App Store Rejection

JakeAi, 2019-02-14:
[Line 692](https://github.com/EddyVerbruggen/nativescript-bluetooth/blob/v2/src/bluetooth.ios.ts#L692) in Bluetooth.ios.ts contains `prefs:root=` which is flagged my the app store review process. The official wording is:

> **Guideline 2.5.1 - Performance - Software Requirements**
Your app uses the "prefs:root=" non-public URL scheme, which is a private entity. The use of non-public APIs is not permitted on the App Store because it can lead to a poor user experience should these APIs change.
> Continuing to use or conceal non-public APIs in future submissions of this app may result in the termination of your Apple Developer account, as well as removal of all associated apps from the App Store.
> **Next Steps**
> To resolve this issue, please revise your app to provide the associated functionality using public APIs or remove the functionality using the "prefs:root" or "App-Prefs:root" URL scheme.

Just a quick search resulted in https://stackoverflow.com/a/54371930

farfromrefug, 2019-02-14:
@JakeAi thanks for reporting. Will fix (remove for now) right away in beta

farfromrefug, 2019-02-14:
done

closed 2020-08-27 by farfromrefug