nativescript-community/ble #133: Android's "cache" refresh

bagir, 2019-03-29:
Greetings, I have a problem with android. 
Android caches information about services and characteristics. 
BLE device can change its characteristics, which leads to communication errors.

Possible solutions described here: https://stackoverflow.com/questions/50739085/how-to-refresh-services-clear-cache

and here:
https://devzone.nordicsemi.com/f/nordic-q-a/10146/refresh-services-in-nrf-master-control-panel

Is it possible to access "refresh()" method of gatt?

farfromrefug, 2019-03-29:
@bagir. ```nativescript-bluetooth``` in its beta version has a discoverAllServices method. Calling it back again would be what you want? Can you try?

bagir, 2019-03-29:
Sorry for offtop, but now I have a problem after npm i --save nativescript-bluetooth@beta with compilation.
**"System.err: Error: com.tns.NativeScriptException: Failed to find module: "events", relative to: app/tns_modules/"**

Cleanup platforms, node-modules, hooks, build with --clean flag and remove application from android is not helped.

If I return npm i --save nativescript-bluetooth@latest error gone..

bagir, 2019-03-29:
discoverAll(DiscoverOptions) didn't work for me:(

closed 2020-12-10 by farfromrefug