nativescript-community/ble #134: Call multiples 'startNotifying' function.

JeffersonLPLima, 2019-04-18:
Hello everyone..
I'm working with a ble glucose meter and  in order to get all the measures I need notify three characteristics:

//Glucose Meter Mandatory Characteristics
var **GLUCOSE_MEASUREMENT_UUID** = "00002a18-0000-1000-8000-00805f9b34fb";
var **GLUCOSE_RECORD_ACCESS_CONTROL_POINT_UUID** = "00002a52-0000-1000-8000-00805f9b34fb";
//Glucose Meter Optional Characteristics
var **GLUCOSE_MEASUREMENT_CONTEXT_UUID** = "00002a34-0000-1000-8000-00805f9b34fb";

So, I already could startNotify each of this three characteristics (by click on one button for each). But I need startNotify automaticaly when I connect to the device. I'm trying call the first `startNotify` function inside of  ` onConnected`  and all the others inside of `.then(function () {` but just the first `startNotify` is executed. May it be a little confuse, but there's a piece of my code:
```
connect_onButtonTap(): void {
            bluetooth.connect({
                UUID: DEVICE_ADDRES,
                onConnected: function (peripheral) {

                    bluetooth.startNotifying({     //only this call is performed
                        ...
                        characteristicUUID: GLUCOSE_MEASUREMENT_UUID,
                        onNotify: function (result) {
                            console.log("GLUCOSE_MEASUREMENT read: " + new Uint8Array(result.value));
                        }
                    }).then(function () {

                        bluetooth.startNotifying({
                            ...
                            characteristicUUID: GLUCOSE_RECORD_ACCESS_CONTROL_POINT_UUID,
                            onNotify: function (result) {
                                console.log("GLUCOSE_RECORD_ACCESS_CONTROL_POINT read: " + new Uint8Array(result.value));
                            }
                        }).then(function () {

                            bluetooth.startNotifying({
                                ...
                                characteristicUUID: GLUCOSE_MEASUREMENT_CONTEXT_UUIDD,
                                onNotify: function (result) {
                                    console.log("GLUCOSE_MEASUREMENT_CONTEXT read: " + new Uint8Array(result.value));
                                }
                            }).then(function () {

                                console.log("GLUCOSE_MEASUREMENT_CONTEXT subscribed for notifications");
                            });
                            console.log("GLUCOSE_RECORD_ACCESS_CONTROL_POINT subscribed for notifications");
                        });
                        console.log("GLUCOSE_MEASUREMENT subscribed for notifications");
                    });

                },
                onDisconnected: function (peripheral) {
                    console.log("Periperhal disconnected with UUID: " + peripheral.UUID);
                }
            });

        }

```
So is there any way to notify all three features as soon as the device is connected?

farfromrefug, 2019-04-18:
use the v2  beta. it allows you to do that?

JeffersonLPLima, 2019-04-19:
Thanks for reply.
Hm.. ok. I'm was with the latest (1.3.1). So I downloaded the latest beta (2.0.0-beta.14), the `p-queue` (5.0), and the `eventemitter3` (3.1.0).

But... I'm got an error when I call the bluetooth.startScanning because  the new version hasn't the commons methods like `startScanning`, `enable`, `startNotify`... 

for the `bluetooth.startScanning` I got...
> LOG from device Galaxy J5: Start scanning
> LOG from device Galaxy J5: An uncaught Exception occurred on "main" thread.
> com.tns.NativeScriptException: 
> Calling js method onClick failed
> 
> TypeError: bluetooth.startScanning is not a function
> File: "file:///data/data/org.nativescript.preview/files/app/bundle.js, line: 270, column: 18
> 
> StackTrace: 
> 	Frame: function:'module.exports.push../home/home-view-model.ts.HomeViewModel.onButtonTap', file:'file:///data/data/org.nativescript.preview/files/app/bundle.js', line: 270, column: 19
> 	Frame: function:'Observable.notify', file:'file:///data/data/org.nativescript.preview/files/app/tns_modules/tns-core-modules/data/observable/observable.js', line: 107, column: 32
> 	Frame: function:'Observable._emit', file:'file:///data/data/org.nativescript.preview/files/app/tns_modules/tns-core-modules/data/observable/observable.js', line: 127, column: 18
> 	Frame: function:'ClickListenerImpl.onClick', file:'file:///data/data/org.nativescript.preview/files/app/tns_modules/tns-core-modules/ui/button/button.js', line: 26, column: 23
> 
> 
> 	at com.tns.Runtime.callJSMethodNative(Native Method)
> 	at com.tns.Runtime.dispatchCallJSMethodNative(Runtime.java:1203)
> 	at com.tns.Runtime.callJSMethodImpl(Runtime.java:1083)
> 	at com.tns.Runtime.callJSMethod(Runtime.java:1070)
> 	at com.tns.Runtime.callJSMethod(Runtime.java:1050)
> 	at com.tns.Runtime.callJSMethod(Runtime.java:1042)
> 	at com.tns.gen.java.lang.Object_button_19_32_ClickListenerImpl.onClick(Object_button_19_32_ClickListenerImpl.java:18)
> 	at android.view.View.performClick(View.java:5716)
> 	at android.widget.TextView.performClick(TextView.java:10926)
> 	at android.view.View$PerformClick.run(View.java:22596)
> 	at android.os.Handler.handleCallback(Handler.java:739)
> 	at android.os.Handler.dispatchMessage(Handler.java:95)
> 	at android.os.Looper.loop(Looper.java:148)
> 	at android.app.ActivityThread.main(ActivityThread.java:7325)
> 	at java.lang.reflect.Method.invoke(Native Method)
> 	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1230)
> 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1120)
> LOG from device Galaxy J5: HMR: Hot Module Replacement Enabled. Waiting for signal.
> LOG from device Galaxy J5: HMR: Hot Module Replacement Enabled. Waiting for signal.
> LOG from device Galaxy J5: HMR: Hot Module Replacement Enabled. Waiting for signal.
> LOG from device Galaxy J5: HMR: Hot Module Replacement Enabled. Waiting for signal.

So, how can I call this methods in this beta version...?  (I'm calling from my `home-view-model.ts`)

JeffersonLPLima, 2019-04-19:
> use the v2 beta. it allows you to do that?

Thanks for reply.
Hm.. ok. I'm was with the latest (1.3.1). So I downloaded the latest beta (2.0.0-beta.14), the `p-queue` (5.0), and the `eventemitter3` (3.1.0).

But... I'm got an error when I call the bluetooth.startScanning because  the new version hasn't the commons methods like `startScanning`, `enable`, `startNotify`... 

for the `bluetooth.startScanning` I got...
> LOG from device Galaxy J5: Start scanning
> LOG from device Galaxy J5: An uncaught Exception occurred on "main" thread.
> com.tns.NativeScriptException:
> Calling js method onClick failed
>
> TypeError: [...]

So, how can I call this methods in this beta version...?  (I'm calling from my `home-view-model.ts`)

farfromrefug, 2019-04-19:
@JeffersonLPLima  yes a lot has changed in V2 . However the typings are correct. So tslint./eslint  should correctly report any error. if that s  still not working there is another issue. what are you using to bundle your app? vue, angular, none?

JeffersonLPLima, 2019-04-19:
> @JeffersonLPLima yes a lot has changed in V2 . However the typings are correct. So tslint./eslint should correctly report any error. if that s still not working there is another issue. what are you using to bundle your app? vue, angular, none?

None... It's just a starter and naive BLE app...  I'm just checking how can I get and use the device data. 

I'm trying to use this [demo of v2](https://github.com/Akylas/nativescript-bluetooth/tree/master/demo). Probably isn't updated on `beta.14` but I'll try anyway.

JeffersonLPLima, 2019-04-21:
@farfromrefug 
While an official ` 2.0` version isn't released, I've been adapted the code to work with the official 1.3.1 using a delay between each call, it is not elegant, but is  stable with my glucose meter.
Thanks for attention, and I'll be watching for new releases!

```
   bluetooth.connect(
        {
            UUID: DEVICE_ADDRES,
            onConnected: function (peripheral) {
                console.log("------- Peripheral connected: " + JSON.stringify(peripheral));

            },
            onDisconnected: function (peripheral) {

            }
        }
    );

    //////
    setTimeout(function () {
        bluetooth.startNotifying({
            peripheralUUID: DEVICE_ADDRES,
            serviceUUID: GLUCOSE_SERVICE_ID_UUID,
            characteristicUUID: GLUCOSE_RECORD_ACCESS_CONTROL_POINT_UUID,
            onNotify: function (result) {
               
            }
        }).then(function () {
            console.log("GLUCOSE_RECORD_ACCESS_CONTROL_POINT subscribed for notifications");
        });
    }, 5000);

    setTimeout(function () {
        bluetooth.startNotifying({
            peripheralUUID: DEVICE_ADDRES,
            serviceUUID: GLUCOSE_SERVICE_ID_UUID,
            characteristicUUID: GLUCOSE_MEASUREMENT_CONTEXT_UUID,
            onNotify: function (result) {
                
            }
        }).then(function () {
            console.log("GLUCOSE_MEASUREMENT_CONTEXTsubscribed for notifications");
        });
    }, 7000);

    //
    setTimeout(function () {
        bluetooth.startNotifying({
            peripheralUUID: DEVICE_ADDRES,
            serviceUUID: GLUCOSE_SERVICE_ID_UUID,
            characteristicUUID: GLUCOSE_MEASUREMENT_UUID,
            onNotify: function (result) {
                
                console.log("GLUCOSE_MEASUREMENT read: " + new Uint8Array(result.value));
            }
        }).then(function () {
            console.log("GLUCOSE_MEASUREMENT subscribed for notifications");
        });
    }, 8000);

    //
    setTimeout(function () {
        bluetooth.write({
            peripheralUUID: DEVICE_ADDRES,
            serviceUUID: GLUCOSE_SERVICE_ID_UUID,
            characteristicUUID: GLUCOSE_RECORD_ACCESS_CONTROL_POINT_UUID,
            value: ['0x01', '0x01'] // a hex 1
        }).then(function (result) {
            console.log("value written");
        }, function (err) {
            console.log("write error: " + err);
        });
    }, 10000);
```

Tyler-V, 2019-07-15:
@JeffersonLPLima were you able to get any readings off bluetooth glucose devices? I am currently trying to read the Contour Next One or Agamatrix device.

JeffersonLPLima, 2019-07-15:
@Tyler-V yep. I already got the measurement reads (and context info) using an Accu Check ble glucose meter. Just run the code above and after 10s the reads are showed in console. Do you already try it?

Tyler-V, 2019-07-15:
@JeffersonLPLima I was able to read the following from a Contour Next One using these values:

```
        const GLUCOSE_SERVICE = '00001808-0000-1000-8000-00805f9b34fb';
        const GLUCOSE_CHARACTERISTIC = '00002a18-0000-1000-8000-00805f9b34fb';
        const RECORDS_CHARACTERISTIC = '00002a52-0000-1000-8000-00805f9b34fb';
        const CONTEXT_CHARACTERISTIC = '00002a34-0000-1000-8000-00805f9b34fb';
```

Which results in the following:

```
JS: ------- _MyGattCallback.onCharacteristicChanged
JS: GLUCOSE_CHARACTERISTIC: 3,0,0,227,7,7,9,17,14,1,16,255,122,176,241
JS: ------- _MyGattCallback.onCharacteristicChanged
JS: GLUCOSE_CHARACTERISTIC: 6,0,1,1
```

I see '122' in the first byte array which is the glucose reading or potentially just a coincidence. No idea what the second value that returns is. Is there a common standard to evaluate these values or how did you read them?

JeffersonLPLima, 2019-07-15:
The 122 (`12th` position of array) is the measurement level. The `6th=day, 7th=hour, 8th=min, 9th=sec`. 
The year is the` 4th << 8 | 3th. `
About the seconde one array (with 4 values), it conrresponds to the context info, like the type or tag of the measurement (e.g. _Pre-prandial, After meat_...). 
About the others positions, I don't know yet... if you discover... please  let me know 😄 .

Tyler-V, 2019-07-15:
That's interesting. I was able to find this which seems to correspond to your findings.

https://github.com/NightscoutFoundation/xDrip/blob/master/app/src/main/java/com/eveningoutpost/dexdrip/GlucoseMeter/GlucoseReadingRx.java

What is also interesting is we're using different meters (accu check / contour next one) and the service / characteristic UUID's are the same. I wonder if this is a common standard among BLE glucose meters?

Were you able to read previous glucose values as well by changing the write value?

JeffersonLPLima, 2019-07-15:
I think that's not possible. The glucose meter with bluetooth 4.0 use this common interface.
Could you send me an email to not change the purpose of this issue?... I'm still working on this and we could talk more about this. [jeffersonlima163@gmail.com](jeffersonlima163@gmail.com)

closed 2020-08-27 by farfromrefug