nativescript-community/ble #135: Not working in Angular projects

solev, 2019-04-23:
I want to use the plugin in a Nativescript Angular project but whenever i call one of this functions
`requestCoarseLocationPermission()` or `hasCoarseLocationPermission()` I get an error:
```
ERROR Error: Uncaught (in promise): Error: java.lang.NullPointerException: Attempt to invoke virtual method 'int android.content.Context.checkPermission(java.lang.String, int, int)' on a null object reference
JS:     android.support.v4.content.ContextCompat.checkSelfPermission(ContextCompat.java:544)
JS:     com.tns.Runtime.callJSMethodNative(Native Method)
JS:     com.tns.Runtime.dispatchCallJSMethodNative(Runtime.java:1203)
JS:     com.tns.Runtime.callJSMethodImpl(Runtime.java:1083)
JS:     com.tns.Runtime.callJSMethod(Runtime.java:1070)
JS:     com.tns.Runtime.callJSMethod(Runtime.java:1050)
JS:     com.tns.Runtime.callJSMethod(Runtime.java:1042)
JS:     com.tns.NativeScriptActivity.onCreate(NativeScriptActivity.java:20)
JS:     android.app.Activity.performCreate(Activity.java:6367)
JS:     android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1110)
JS:     android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2404)
JS:     android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2511)
JS:     android.app.ActivityTh...
```
Using Nativescript 5 and Angular 7.
Tested on Android 6 and Android 7.

Note: I've tried to use it in a normal typescript project (non angular) and it works fine

erikbjork, 2019-04-26:
Move the function calls to a function triggered by the loaded event on Page:

```
<Page (loaded)="onPageLoaded()">
   ...
</Page>
```

```
onPageLoaded() {
  hasCoarseLocationPermission()
}
```

solev, 2019-09-18:
Thanks for that. A little bit late, but closing since the provided solution is working

closed 2019-09-18 by solev