nativescript-community/ble #136: Unable to run ng-demo in v2 (beta)

drewcovi, 2019-04-25:
For some reason the ng-demo in v2 crashes and burns when deploying with the message

`NativeScript encountered a fatal error: SyntaxError: Importing binding name 'Observable' is not found.`

farfromrefug, 2019-04-25:
please send the full error log

drewcovi, 2019-04-25:
hey @farfromrefug! thanks for the response. 

Here's the complete log:

```
Searching for devices...
Your application will be deployed only on the device specified by the provided index or identifier.
Executing before-liveSync hook from /Users/h101426/Github/nativescript-bluetooth/demo-ng/hooks/before-liveSync/nativescript-angular-sync.js
Executing before-watchPatterns hook from /Users/h101426/Github/nativescript-bluetooth/demo-ng/hooks/before-watchPatterns/nativescript-dev-sass.js
Executing before-watchPatterns hook from /Users/h101426/Github/nativescript-bluetooth/demo-ng/hooks/before-watchPatterns/nativescript-dev-typescript.js
Executing before-watchPatterns hook from /Users/h101426/Github/nativescript-bluetooth/demo-ng/hooks/before-watchPatterns/nativescript-dev-webpack.js
Executing before-watch hook from /Users/h101426/Github/nativescript-bluetooth/demo-ng/hooks/before-watch/nativescript-dev-sass.js
Found peer node-sass
No input file was found.

SASS compiler failed with exit code 1
Executing before-watch hook from /Users/h101426/Github/nativescript-bluetooth/demo-ng/hooks/before-watch/nativescript-dev-typescript.js
Found peer TypeScript 3.1.6
Executing before-watch hook from /Users/h101426/Github/nativescript-bluetooth/demo-ng/hooks/before-watch/nativescript-dev-webpack.js
Executing before-shouldPrepare hook from /Users/h101426/Github/nativescript-bluetooth/demo-ng/hooks/before-shouldPrepare/nativescript-dev-webpack.js
Executing before-prepare hook from /Users/h101426/Github/nativescript-bluetooth/demo-ng/hooks/before-prepare/nativescript-dev-sass.js
Hook skipped because either bundling or livesync is in progress.
Executing before-prepare hook from /Users/h101426/Github/nativescript-bluetooth/demo-ng/hooks/before-prepare/nativescript-dev-typescript.js
Hook skipped because either bundling or livesync is in progress.
Preparing project...
Executing before-prepareJSApp hook from /Users/h101426/Github/nativescript-bluetooth/demo-ng/hooks/before-prepareJSApp/nativescript-dev-webpack.js
Project successfully prepared (iOS)
Executing after-prepare hook from /Users/h101426/Github/nativescript-bluetooth/demo-ng/hooks/after-prepare/nativescript-dev-sass.js
Executing after-prepare hook from /Users/h101426/Github/nativescript-bluetooth/demo-ng/hooks/after-prepare/nativescript-dev-webpack.js
Refreshing application on device 00008020-001E450A3E98002E...
Unzipping LiveSync folder. This could take a while...
Successfully synced application com.drewcovi.bluetoothng on device 00008020-001E450A3E98002E.
Unzipped 6317 entries in 4158.501029ms.
***** Fatal JavaScript exception - application has been terminated. *****
Native stack trace:
1   0x10098d194 NativeScript::reportFatalErrorBeforeShutdown(JSC::ExecState*, JSC::Exception*, bool, bool)
2   0x1009c3720 -[TNSRuntime executeModule:referredBy:]
3   0x100010538 main
4   0x21037cffc <redacted>
JavaScript stack trace:
1   moduleDeclarationInstantiation@file:///app/services/bluetooth.service.js:7:39
2   link@[native code]
3   link@[native code]
4   linkAndEvaluateModule@[native code]
5   promiseReactionJob@[native code]
6   require@[native code]
7   anonymous@file:///app/services/bluetooth.service.js:7:39
8   evaluate@[native code]
9   moduleEvaluation@[native code]
10  promiseReactionJob@[native code]
11  require@[native code]
12  anonymous@file:///app/components/scan.component/scan.component.js:5:34
13  evaluate@[native code]
14  moduleEvaluation@[native code]
15  promiseReactionJob@[native code]
16  require@[native code]
17  anonymous@file:///app/app.routing.js:4:31
18  evaluate@[native code]
19  moduleEvaluation@[native code]
20  promiseReactionJob@[native code]
21  require@[native code]
22  anonymous@file:///app/app.module.js:7:28
23  evaluate@[native code]
24  moduleEvaluation@[native code]
25  promiseReactionJob@[native code]
26  require@[native code]
27  anonymous@file:///app/main.js:5:27
28  evaluate@[nati<\M-b\M^@\M-&>
JavaScript error:
file:///app/services/bluetooth.service.js:7:39: JS ERROR SyntaxError: Importing binding name 'Observable' is not found.
*** JavaScript call stack:
(
)
*** Terminating app due to uncaught exception 'NativeScript encountered a fatal error: SyntaxError: Importing binding name 'Observable' is not found.
at
1   moduleDeclarationInstantiation@file:///app/services/bluetooth.service.js:7:39
2   link@[native code]
3   link@[native code]
4   linkAndEvaluateModule@[native code]
5   promiseReactionJob@[native code]
6   require@[native code]
7   anonymous@file:///app/services/bluetooth.service.js:7:39
8   evaluate@[native code]
9   moduleEvaluation@[native code]
10  promiseReactionJob@[native code]
11  require@[native code]
12  anonymous@file:///app/components/scan.component/scan.component.js:5:34
13  evaluate@[native code]
14  moduleEvaluation@[native code]
15  promiseReactionJob@[native code]
16  require@[native code]
17  anonymous@file:///app/app.routing.js:4:31
18  evaluate@[native code]
19  moduleEvaluation@[native code]
20  promiseReactionJob@[native code]
21  require@[native code]
22  anonymous@file:///app/app.module.js:7:28
23  evaluate@[native code]
<\M-b\M^@\M-&>
Executing after-watch hook from /Users/h101426/Github/nativescript-bluetooth/demo-ng/hooks/after-watch/nativescript-dev-sass.js
Executing after-watch hook from /Users/h101426/Github/nativescript-bluetooth/demo-ng/hooks/after-watch/nativescript-dev-typescript.js
Stopping tsc watch
Executing after-watch hook from /Users/h101426/Github/nativescript-bluetooth/demo-ng/hooks/after-watch/nativescript-dev-webpack.js
Stopping webpack watch
```

farfromrefug, 2019-04-25:
@drewcovi please pull and try again (make sure the plugin is compile again with tsc)

drewcovi, 2019-04-25:
@farfromrefug thanks for the quick update! 

Seems it wont start now with the following error:

```
***** Fatal JavaScript exception - application has been terminated. *****
Native stack trace:
1   0x104dc9194 NativeScript::reportFatalErrorBeforeShutdown(JSC::ExecState*, JSC::Exception*, bool, bool)
2   0x104df4518 NativeScript::FFICallback<NativeScript::ObjCMethodCallback>::ffiClosureCallback(ffi_cif*, void*, void**, void*)
3   0x1058521c8 ffi_closure_SYSV_inner
4   0x1058541b4 .Ldo_closure
5   0x2113aa3bc <redacted>
6   0x2108a2070 <redacted>
7   0x2108a2030 <redacted>
8   0x2108a14d4 <redacted>
9   0x2108a1160 <redacted>
10  0x210818488 <redacted>
11  0x2108a0c08 _CFXNotificationPost
12  0x21129bcfc <redacted>
13  0x23d80c9c4 <redacted>
14  0x23d8122b0 <redacted>
15  0x23d0a5e20 <redacted>
16  0x23d0ae7e8 <redacted>
17  0x23d0a5a88 <redacted>
18  0x23d0a63a4 <redacted>
19  0x23d0a45ec <redacted>
20  0x23d0a42b0 <redacted>
21  0x23d0a8fd0 <redacted>
22  0x23d0a9e1c _performActionsWithDelayForTransitionContext
23  0x23d0a8e84 <redacted>
24  0x23d0ade68 <redacted>
25  0x23d8107ec <redacted>
26  0x23d3f3648 <redacted>
27  0x2132f13a8 <redacted>
28  0x2132fb6e4 <redacted>
29  0x2132fae34 <redacted>
30  0x2103458f4 <redacted>
31  0x210348ecc <redacted>
JavaScript stack trace:
1   resolve@file:///app/tns_modules/nativescript-angular/resource-loader.js:33:103
2   get@file:///app/tns_modules/nativescript-angular/resource-loader.js:19:40
3   _fetch@file:///app/tns_modules/@angular/compiler/bundles/compiler.umd.js:15935:50
4   _preParseTemplate@file:///app/tns_modules/@angular/compiler/bundles/compiler.umd.js:15974:39
5   normalizeTemplate@file:///app/tns_modules/@angular/compiler/bundles/compiler.umd.js:15962:57
6   loadDirectiveMetadata@file:///app/tns_modules/@angular/compiler/bundles/compiler.umd.js:18154:79
7   @file:///app/tns_modules/@angular/compiler/bundles/compiler.umd.js:25778:80
8   forEach@[native code]
9   @file:///app/tns_modules/@angular/compiler/bundles/compiler.umd.js:25777:83
10  forEach@[native code]
11  _loadModules@file:///app/tns_modules/@angular/compiler/bundles/compiler.umd.js:25774:86
12  _compileModuleAndComponents@file:///app/tns_modules/@angular/compiler/bundles/compiler.umd.js:25752:52
13  compileModuleAsync@file:///app/tns_modules/@angular/compil<\M-b\M^@\M-&>
JavaScript error:
file:///app/tns_modules/nativescript-angular/resource-loader.js:33:103: JS ERROR Error: Could not resolve actionBar.component.html. Looked for: /var/mobile/Containers/Data/Application/FEC7E952-3082-449A-9E2F-99D54DC6C11F/Library/Application Support/LiveSync/app/actionBar.component.html.
*** JavaScript call stack:
(
0   UIApplicationMain@[native code]
1   start@file:///app/tns_modules/tns-core-modules/application/application.js:275:26
2   run@file:///app/tns_modules/tns-core-modules/application/application.js:303:10
3   bootstrapNativeScriptApp@file:///app/tns_modules/nativescript-angular/platform-common.js:187:26
4   bootstrapApp@file:///app/tns_modules/nativescript-angular/platform-common.js:105:38
5   bootstrapModule@file:///app/tns_modules/nativescript-angular/platform-common.js:89:26
6   anonymous@file:///app/main.js:6:57
7   evaluate@[native code]
8   moduleEvaluation@:1:11
9   promiseReactionJob@:1:11
)
*** Terminating app due to uncaught exception 'NativeScript encountered a fatal error: Error: Could not resolve actionBar.component.html. Looked for: /var/mobile/Containers/Data/Application/FEC7E952-3082-449A-9E2F-99D54DC6C11F/Library/Application Support/LiveSync/app/actionBar.component.html.
at
1   resolve@file:///app/tns_modules/nativescript-angular/resource-loader.js:33:103
2   get@file:///app/tns_modules/nativescript-angular/resource-loader.js:19:40
3   _fetch@file:///app/tns_modules/@angular/compiler/bundles/compiler.umd.js:15935:50
4   _preParseTemplate@file:///app/tns_modules/@angular/compiler/bundles/compiler.umd.js:15974:39
5   normalizeTemplate@file:///app/tns_modules/@angular/compiler/bundles/compiler.umd.js:15962:57
6   loadDirectiveMetadata@file:///app/tns_modules/@angular/compiler/bundles/compiler.umd.js:18154:79
7   @file:///app/tns_modules/@angular/compiler/bundles/compiler.umd.js:25778:80
8   forEach@[native code]
9   @file:///app/tns_modules/@angular/compiler/bundles/compiler.umd.<\M-b\M^@\M-&>
```

farfromrefug, 2019-04-26:
better this has nothing to do with the Bluetooth module itself. can you try and find out what it is? dont really have tile to try and run it right now

cowboyd, 2019-05-06:
@farfromrefug Don't know if it's related, but we definitely saw the Android runtime break when upgrading from `1.3.1w to `beta.18` https://github.com/resideo-platform/exemplar/issues/118

I'm happy to help debug if there's anything that I can try out. I've filed a corresponding issue on NativeScript https://github.com/NativeScript/NativeScript/issues/7205

If you'd like I can file a separate issue on this repository, but not sure which you'd prefer.

farfromrefug, 2019-05-06:
Your issue has nothing to do with bluetooth or the runtime. This is simply a build issue. What you did by updating the runtime is actually clean all and run again. This is something i see quite often.

cowboyd, 2019-05-06:
> Your issue has nothing to do with bluetooth or the runtime. This is simply a build issue. What you did by updating the runtime is actually clean all and run again. This is something i see quite often.

There may be something else going on here, but this hypothesis appears to be inconsistent with our observations.

To reproduce:

1. checkout this commit resideo-platform/exemplar@7327b0fc5390ee5f13e92d9e8e6d96eff26004ce which uses `nativescript-bluetooth
@1.3.1`
1. `yarn install`
1. `yarn start android` confirm that application builds and loads
1. cherry pick resideo-platform/exemplar@61ddaed23002b9e1478836d854decb77b4643f51 which does nothing else except upgrade nativescript-bluetooth
1. *without cleaning* any build artifacts, run `yarn install`
1. `yarn start android` confirm that application builds, but fails to load
1. perform a hard reset to the original commit resideo-platform/exemplar@7327b0fc5390ee5f13e92d9e8e6d96eff26004ce
1. *without cleaning* any build artifacts, run `yarn install`
1. `yarn start android` confirm that applications builds, and successfully loads.

It may very well be a build issue, but this evidence suggests very strongly that there is some interaction between this commit  resideo-platform/exemplar@61ddaed23002b9e1478836d854decb77b4643f51 and the android build not happening properly since upgrading without cleaning causes it to fail, and also downgrading without cleaning causes it to succeed again.

It could also be something to do with `eventemitter3` and `p-queue` which are the transient dependencies of `nativescript-bluetooth`?

> Or Maybe you update from 1.3.1 to beta18 without cleaning everything? That would explain it.

Our CI environment builds from scratch every time, and we are seeing this there as well.

I'm not super familiar with the NativeScript build process, but it appears that there is something going on here since the only variable that we're changing is the bluetooth plugin.

farfromrefug, 2019-05-07:
Your 9 steps dont' actually say anything interesting. The hard reset obviously work as you end up with a "corresponding" node_modules/build folder.
What you need to do is;
-1,2,3,4,5 =>fail
- remove the platforms folder
- ```yarn start android```
Then see what happens. Also you can post the full build/run logs. That might help

cowboyd, 2019-05-07:
I did steps 1,2,3,4,5.

- removed platforms folder
- yarn start android

and here is the build and devices log https://gist.github.com/cowboyd/1852c1df3484c66748f8aeaf38c257b0#file-3-buildlog-clean-ns-bluetooth-2_0_0_beta_18

farfromrefug, 2019-05-09:
@cowboyd i don't see anything wrong there :s Hope the {N} team can help you tracking the issue. It usually is related to native classes extended through JS. We do that in the ```nativescript-bluetooth``` ! Though i cant see where it fails seeing your logs. Really sorry i can't help more there

cowboyd, 2019-05-09:
Yeah, it's a real head-scratcher. Thanks for your time, and I'll circle back with them to see if they can provide any insight.

cowboyd, 2019-05-10:
It looks like there is a purported fix https://github.com/NativeScript/android-runtime/pull/1374 although I'm not sure what it is fixing.

closed 2020-08-27 by farfromrefug