nativescript-community/ble #137: optimistic connect attempts?

bniehuser, 2019-04-27:
is it possible to attempt to connect to a peripheral by UUID without first scanning to see if it's in range?  the app only needs to connect to a specific peripheral, which has already been identified by UUID (and mac address, service uuids, etc) -- i would like to be able to attempt connection on an interval without having to initiate a scan process each time to test its availability.  When i try i get `file:///app/tns_modules/nativescript-bluetooth/bluetooth.js:463:18: Error in Bluetooth.connect: TypeError: null is not an object (evaluating 'Bluetooth._state.peripheralArray.count')`, presumably because there are no scan results to check against.  I'm using the nativescript + angular (typescript) with the current release, 1.3.1
not even sure this is possible, but it was suggested to me as a way to auto-reconnect to a known device and avoid the overhead of 1-2 second device scans on an endless 5-10 second interval, and figured i'd check with the experts ;).

farfromrefug, 2019-04-30:
@bniehuser should already be possible in the 2.0 beta. just use connect and your UUID in the params

bniehuser, 2019-04-30:
absolutely works, at least for iOS.  just throw a connect attempt on app load at whatever device you want connected, and another in the disconnect handler.  boom; auto-reconnect.

farfromrefug, 2019-05-01:
@bniehuser  so it works? could you send a log for the thrown error?