nativescript-community/ble #140: onDiscovered method is not calling in my app

BoobalakannanS, 2019-06-04:
In my app, I am trying to connect with Bluetooth devices using the code given below
` bluetooth.isBluetoothEnabled().then(
    function (enabled) {
        console.log("Enabled? " + enabled);
        if (enabled) {
            bluetooth.startScanning({
                serviceUUIDs: [],
                seconds: 9,
                onDiscovered: function (peripheral) {
                    console.log("Periperhal found with UUID: " + peripheral.UUID);
                },
                skipPermissionCheck: true,
            }).then(function () {
                console.log("scanning complete");
            }, function (err) {
                console.log("error while scanning: " + err);
            });
        } else {
            this.requestBluetooth();
        }
    }
);
In this, `onDiscovered method is not calling. But, it is hitting in "then()".
Please give me the solution to solve this. Thanks in advance!

EddyVerbruggen, 2019-06-12:
Hi, thanks for reporting this issue.

Are you connecting to a Bluetooth 4 device (also known as BLE/Bluetooth Low Energy)? Because that's what the plugin is for.

BoobalakannanS, 2019-06-14:
Dear EddyVerbruggen,
Thanks for your reply. Yes, I tried to connect with a Bluetooth 4 mobile device.

EddyVerbruggen, 2019-06-14:
Can you please provide a link to the device?

If it’s not too costly I may purchase it and use it as a testdevice myself.

BoobalakannanS, 2019-06-17:
I tried with two [Redmi 4A](https://www.mi.com/in/redmi4a/specs/) mobiles . But I can't see the available devices in onDiscovered method.

EddyVerbruggen, 2019-06-17:
@BoobalakannanS Are those devices Bluetooth peripherals or phones? I mean, this plugin is intended to be used on a phone/tablet and connect to Bluetooth Peripherals. You can't connect to other phones afaik.

To save us all some time it would help to learn your actual use case. If it's connecting from a phone (aka "Central") to a BLE client device (aka "peripheral", which broadcasts its services) then this is the plugin for you. Otherwise it's not.

Also, connecting to a Bluetooth peripheral of a lower spec version than 4.0 like https://www.amazon.in/gp/product/B0714CX63G/ref=ppx_yo_dt_b_asin_title_o03_s00?ie=UTF8&psc=1 and https://www.zebra.com/content/dam/zebra_new_ia/en-us/solutions-verticals/product/RFID/Reader/rfd8500-handheld-rfid-reader/spec-sheet/RFD8500-Specification-Sheet-A4.pdf will not work with this plugin.

BoobalakannanS, 2019-06-18:
Thank you @EddyVerbruggen .  In which device will this plugin work? could you please refer me any Bluetooth peripherals which will work with this plugin.

EddyVerbruggen, 2019-06-18:
For instance devices like a [Polar heartrate sensor](https://www.polar.com/us-en/products/heart-rate-sensors) or the [Texas Instruments SensorTag](https://evothings.com/things/texasinstruments). Really any BLE Peripheral should work.

closed 2019-07-08 by EddyVerbruggen