nativescript-community/ble #142: cannot communicate with devices if the phone remembers them (ios)

JoshuaMcCoy, 2019-07-16:
On iOS.  I cannot access the device on the second run of the app.

When I connect to a device, IOS remembers it. and I i cannot talk to remembered devices. if I enter the settings, and select 'forget device' then I can once again connect to it, but closing and re-opening the app always results in the device being added to remembered devices, and I have to select'forget device' before running my app every time.

farfromrefug, 2019-07-16:
@JoshuaMcCoy this is not an issue with this module. This is mostly because you forgot to disconnect from the device. On app close make sure you disconnect from your device

JoshuaMcCoy, 2019-07-16:
disconnecting before close does not resolve the issue. I can do the following:

scan() -> device shows up
connect(uuid) -> console says 'connected successfully'
disconnect(uuid) -> 'disconnected successfully
close app
open settings -> bluetooth -> device says connected
open app
scan() -> device does not appear in scan result
connect(uuid) -> console says 'Could not find peripheral with UUID'
write(uuid) -> 'Could not find peripheral with UUID'

open settings --> bluetooth -> forget device
return to app, run connect(uuid) -> 'successfully connected.'

closed 2020-08-27 by farfromrefug