nativescript-community/ble #143: Error on android when calling startScanning()

JoshuaMcCoy, 2019-07-17:
when I call startScanning() on android, I get the following error:
`[moto x4]: ERROR TypeError: Cannot read property 'scan' of undefined
[moto x4]: Error in Bluetooth.startScanning: TypeError: Cannot read property 'ContextCompat' of undefined
[moto x4]: error while scanning: TypeError: Cannot read property 'ContextCompat' of undefined`

it even happens when the code is exactly as recommended in the example:
`bluetooth.startScanning({
  serviceUUIDs: [],
  seconds: 4,
  onDiscovered: function (peripheral) {
  	console.log("Periperhal found with UUID: " + peripheral.UUID);
  },
  skipPermissionCheck: false,
}).then(function() {
  console.log("scanning complete");
}, function (err) {
  console.log("error while scanning: " + err);
});`

Exxili, 2019-07-22:
Having Same issue - ContextCompat Error as well

EddyVerbruggen, 2019-07-22:
That’s because N6 uses AndroidX. I’ve got a local fix but not entirely sure about what to release at this point: master or the v2 branch. Should be more clear soon..

JoshuaMcCoy, 2019-07-23:
@EddyVerbruggen 
Thanks for creating this useful package! Is the local fix you have in place straight forward enough for you to point me at the solution? Dont worry about it if it will take you away from working on your commit.

toams69, 2019-08-25:
Having Same issue - ContextCompat Error as well

`TypeError: Cannot read property 'ContextCompat' of undefined
JS:     at Object.push.../node_modules/nativescript-bluetooth/bluetooth.js.Bluetooth._coarseLocationPermissionGranted file:///node_modules/nativescript-bluetooth/bluetooth.js:18:35
JS:     at file:///node_modules/nativescript-bluetooth/bluetooth.js:517:57
JS:     at new Promise (<anonymous>)
JS:     at Object.push.../node_modules/nativescript-bluetooth/bluetooth.js.Bluetooth.startScanning file:///node_modules/nativescript-bluetooth/bluetooth.js:444:9

JoshuaMcCoy, 2019-08-28:
upgrading to nativescriptbluetooth@beta can resolve this issue.  there are some differences in the way various methods work in the beta version, but they are minor and i was able to use intelesense to figure out the changes.

to install:
`npm install nativescript-bluetooth@beta`

to use, I declare it like this:
`import { Bluetooth } from 'nativescript-bluetooth';
const bluetooth = new Bluetooth();`

schnapzz, 2019-09-20:
Had the same issue, fixed it by changing the following in the built node_modules/nativescript-bluetooth/bluetooth.android.js, line no. ~18

`android.support.v4.content.ContextCompat.checkSelfPermission(application.android.foregroundActivity, android.Manifest.permission.ACCESS_COARSE_LOCATION);
`
to 
`androidx.core.content.ContextCompat.checkSelfPermission(application.android.foregroundActivity, android.Manifest.permission.ACCESS_COARSE_LOCATION);`

This is following the new class mappings to androidx from the Android Dev team [here](https://developer.android.com/jetpack/androidx/migrate/class-mappings). This is by no means a solid solution but for my project it is sufficient untill a patch to the plugin is live.

StephaniesHusband, 2019-10-21:
When i use the @beta version Android starts working but the iOS version does not work anymore. Anyone else seeing this?

methompson, 2020-03-09:
I know this hasn't been resolved yet, but I took the suggestion of @schnapzz and placed the following in my NS-Vue app to bridge compatibility. I check that android.support.v4 exists in the global object, then copy a reference to android.core.content.ContextCompat into android.support.v4.content.ContextCompat.

I don't believe this is a particularly good solution, but it's workable without actually modifying any modules.

```
// in app.js
import { isAndroid } from "tns-core-modules/platform";

if (isAndroid) {
  if ( android
    && android.support
    && android.support.v4
  ) {
    android.support.v4.content = {
      ContextCompat: androidx.core.content.ContextCompat,
    };
  }
}
```

closed 2020-08-27 by farfromrefug