nativescript-community/ble #145: Does not work with Nativescript 6.0

msn444, 2019-07-19:
Calling startScanning() with Nativescript 6.0 (Clean vue base app) produces the following error:

`JS: 'Error in Bluetooth.startScanning: TypeError: Cannot read property \'ContextCompat\' of undefined'`

```
│ Plugin                  │ Version │
│ nativescript-bluetooth  │ ^1.3.1  │
│ nativescript-theme-core │ ~1.0.6  │
│ nativescript-vue        │ ~2.3.0  │
│ tns-core-modules        │ ~6.0.0  │
```

Tyler-V, 2019-07-19:
Did you migrate your project? 

`tns migrate`

msn444, 2019-07-19:
I created it in NS 6.0. I don’t think there’s anything to migrate is there?

msn444, 2019-07-19:
nativescript-bluetooth-2.0.0 seems to work though.

farfromrefug, 2019-07-20:
@msn444 indeed it does

zhiyong80, 2019-07-22:
Got error when building the demo app:
```
cd demo
npm install
tns migrate
tns build android --bundle
```
Unknown argument: preserve-symlinks
Executing webpack failed with exit code 1.

MeenalBahua, 2019-07-30:
How to migrate to 2.0.0?Can someone please provide steps.

msn444, 2019-07-30:
These are the migration details I've discovered so far:

- It's a class now, so you need
`
import { Bluetooth } from 'nativescript-bluetooth';
const bluetooth = new Bluetooth();
`
- the `serviceUUID` property that was formerly a child of
the`startScanning` parameter set is now contained inside a `filters` object
- `write` now requires `value` to be passed as a byte array rather than a
string of comma-separated hex values

msn444, 2019-07-30:
Sorry, formatting didn't work:

- It's a class now, so you need
`
import { Bluetooth } from 'nativescript-bluetooth';
const bluetooth = new Bluetooth();
`
- the `serviceUUID` property that was formerly a child of
the`startScanning` parameter set is now contained inside a `filters` object
- `write` now requires `value` to be passed as a byte array rather than a
string of comma-separated hex values

MeenalBahua, 2019-07-31:
It worked. Thank you.

mcvtec, 2019-08-18:
@msn444 I am facing the same error, could you explain how to update the module?

MeenalBahua, 2019-08-19:
used the beta version of the this lib and followed the example for the same.

closed 2020-08-27 by farfromrefug