nativescript-community/ble #146: NativeScript 6 + nativescript-bluetooth-v2.0.0-beta doesn't work

RyanH1234, 2019-07-24:
Recently I upgraded to NativeScript 6 - I came across the error : 

` JS: 'Error in Bluetooth.startScanning: TypeError: Cannot read property \'ContextCompat\' of `undefined'````

I found the apparent solution to upgrade to the nativescript-bluetooth beta via 
 
`npm i install nativescript-bluetooth@beta`

My package.json file has been altered like so 

`"dependencies": {
"nativescript-bluetooth": "^2.0.0-beta.23"
}` 

However whenever I run `tns run android` I get the error 

`Cannot read property 'isBluetoothEnabled' of undefined`

Any thoughts to why this might be happening?

farfromrefug, 2019-07-24:
Can you share your package.json

RyanH1234, 2019-07-24:
Sure!

> {
>   "nativescript": {
>     "id": "org.nativescript.application",
>     "tns-ios": {
>       "version": "6.0.1"
>     },
>     "tns-android": {
>       "version": "6.0.0"
>     }
>   },
>   "dependencies": {
>     "install": "^0.13.0",
>     "nativescript-bluetooth": "^2.0.0-beta.23",
>     "nativescript-cardview": "^3.2.0",
>     "nativescript-vue": "^2.0.0",
>     "tns-core-modules": "6.0.1",
>     "vuex": "^3.0.1"
>   },
>   "devDependencies": {
>     "@babel/core": "^7.0.0",
>     "@babel/preset-env": "^7.0.0",
>     "babel-loader": "^8.0.2",
>     "babel-traverse": "6.26.0",
>     "babel-types": "6.26.0",
>     "babylon": "6.18.0",
>     "clean-webpack-plugin": "^0.1.19",
>     "copy-webpack-plugin": "^4.5.2",
>     "css-loader": "^1.0.0",
>     "lazy": "1.0.11",
>     "nativescript-dev-webpack": "1.0.1",
>     "nativescript-vue-template-compiler": "^2.0.0",
>     "nativescript-worker-loader": "~0.9.0",
>     "node-sass": "^4.9.2",
>     "sass-loader": "^7.1.0",
>     "terser-webpack-plugin": "^1.1.0",
>     "vue-loader": "^15.2.6",
>     "webpack": "^4.16.4",
>     "webpack-bundle-analyzer": "~2.13.1",
>     "webpack-cli": "^3.1.0"
>   }
> }
>

farfromrefug, 2019-07-25:
@RyanH1234 what device are you running this on? What os version?

RyanH1234, 2019-07-25:
Samsung A5 running Android 8.0.0

farfromrefug, 2019-07-25:
@RyanH1234 Can you share the full log of the error? need to know where it happens in the code

RyanH1234, 2019-07-25:
The logs say "project successfully built", the elements are all rendered and then I run into this....

> System.err: An uncaught Exception occurred on "main" thread.
> System.err: Calling js method run failed
> System.err: TypeError: Cannot read property 'isBluetoothEnabled' of undefined
> System.err:
> System.err: StackTrace:
> System.err:     Frame: function:'', file:'file:///app\pages\Connection.vue:53:0
> System.err:     Frame: function:'invoke', file:'file:///node_modules\tns-core-modules\timer\timer.js:52:47
> System.err:     Frame: function:'run', file:'file:///node_modules\tns-core-modules\timer\timer.js:56:12
> System.err:     at com.tns.Runtime.callJSMethodNative(Native Method)
> System.err:     at com.tns.Runtime.dispatchCallJSMethodNative(Runtime.java:1242)
> System.err:     at com.tns.Runtime.callJSMethodImpl(Runtime.java:1122)
> System.err:     at com.tns.Runtime.callJSMethod(Runtime.java:1109)
> System.err:     at com.tns.Runtime.callJSMethod(Runtime.java:1089)
> System.err:     at com.tns.Runtime.callJSMethod(Runtime.java:1081)
> System.err:     at com.tns.gen.java.lang.Runnable.run(Runnable.java:17)
> System.err:     at android.os.Handler.handleCallback(Handler.java:789)
> System.err:     at android.os.Handler.dispatchMessage(Handler.java:98)
> System.err:     at android.os.Looper.loop(Looper.java:164)
> System.err:     at android.app.ActivityThread.main(ActivityThread.java:6944)
> System.err:     at java.lang.reflect.Method.invoke(Native Method)
> System.err:     at com.android.internal.os.Zygote$MethodAndArgsCaller.run(Zygote.java:327)
> System.err:     at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1374)

farfromrefug, 2019-07-25:
@RyanH1234  Do you know if that ```isBluetoothEnabled``` call is from your code or you don't ever call ```isBluetoothEnabled```?
If you do call it, can you share the code? (i need to see the call and the creating of the object you are calling this on).

RyanH1234, 2019-07-25:
So I'm using nativescript-vue : I import the plugin like so

> import bluetooth from "nativescript-bluetooth"; 

and call the following 

> bluetooth.isBluetoothEnabled().then(enabled => {
> // logic 
> });

Within the mounted() function

The imported "bluetooth" is null - hence the error

I saw somewhere that importing plugins has changed in NativeScript 6 - that you need to use absolute paths, not sure what the absolute path would be?

> import bluetooth from 'node_modules/nativescript-blueooth'

The above doesn't work either

farfromrefug, 2019-07-25:
@RyanH1234 That explains it!
do it like this for now. will fix the beta:
```typescript
import { Bluetooth } from 'nativescript-bluetooth';
const bluetooth = new Bluetooth();
...
bluetooth.isBluetoothEnabled()
```

RyanH1234, 2019-07-25:
@farfromrefug It works! Thanks so much!

firumon, 2019-08-31:
dear @farfromrefug, my device is not discovering any near by devices.. Following the same NS6+2.0.0-beta.23.. 

> "dependencies": {
    "install": "^0.13.0",
    "nativescript-bluetooth": "^2.0.0-beta.23",
    "nativescript-theme-core": "~1.0.6",
    "nativescript-vue": "~2.4.0",
    "tns-core-modules": "~6.0.0"
  },
Only NativeScript App is not discovering, **native bluetooth scan discovering**, how come this??
What I am doing is, created fresh NS-Vue install using sidekick, installed `npm i install nativescript-bluetooth@beta`, then added following lines in app.js

> import { Bluetooth } from 'nativescript-bluetooth';
global.bluetooth = new Bluetooth();

then, i run it on my real device (ASUS Zenfone Max M1 Pro - Android 9) using sidekick, enabled debugger.. Once installed on device, i tested methods like `isBluetoothEnabled`,`hasCoarseLocationPermission` etc etc, all working as per documentation, but `startScanning->onDiscovered`  not working.. 

**Please help, I am in the last phase of  a project, without this I can't move a bit forward!!**

closed 2020-08-27 by farfromrefug