nativescript-community/ble #147: peripheral_not_connected

dearpriyankasa, 2019-08-02:
Hello, I am getting a "peripheral_not_connected" error when I am trying to write to the device or read from the device. When I display the peripheral object itself it shows "state":"connected". The problem arises when I try to read to write to the device. 
I am using NativeScript 6.0.2 and nativescript-bluetooth plugin ^2.0.0-beta.23.
npm version is 6.9.0.

farfromrefug, 2019-08-02:
@dearpriyankasa can i get the error callstack please.

dearpriyankasa, 2019-08-02:
Here is my code snippet for read and write. It goes in the error section of read and the error message is fed to feedback variable and displayed using HTML on screen.

public onCharacteristicsTap(args) {

    var characteristic = args;
    let array: number[] = [32,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0];
    const p = characteristic.properties;
    const actions = [];

    if (p.read) actions.push("read");
    if (p.write) actions.push("write");
    if (p.write) actions.push("write 0x01"); // convenience method, will write hex 1, translated to a binary 1
    if (p.write) actions.push("write bytearray");
    if (p.writeWithoutResponse) actions.push("writeWithoutResponse");
    if (p.notify) actions.push("notify start");
    if (p.notify) actions.push("notify stop");

    dialogs
      .action({
        message: "Pick a property",
        cancelButtonText: "Cancel",
        actions: actions
      })
      .then(result => {
        if (result === "read") {
        this._bluetooth
            .read({
              peripheralUUID: this.peripheral.UUID,
              serviceUUID: this.service.UUID,
              characteristicUUID: characteristic.UUID
            })
            .then(
              result => {
                var data = new Uint8Array(result.value);
                this.feedback = JSON.stringify(data);
                this.feedbackTimestamp = this._getTimestamp();
              },
              error => {
                console.log("Peripheral is:",this.peripheral);
                this.feedback = error;
                this.feedbackTimestamp = this._getTimestamp();
              }
            );
        } else if (result === "write") {
          dialogs
            .prompt({
              title: "Write what exactly?",
              message:
                "Please enter byte values; use 0x in front for hex and send multiple bytes by adding commas. For example 0x01 or 0x007F,0x006E",
              cancelButtonText: "Cancel",
              okButtonText: "Write it!"
            })
            .then(response => {
              if (response.result) {
                this._bluetooth
                  .write({
                    peripheralUUID: this.peripheral.UUID,
                    serviceUUID: this.service.UUID,
                    characteristicUUID: characteristic.UUID,
                    value: response.text
                  })
                  .then(
                    result => {
                        this.feedback = "value written";
                        this.feedbackTimestamp = this._getTimestamp();
                    },
                    errorMsg => {
                        this.feedback = errorMsg;
                        this.feedbackTimestamp = this._getTimestamp();
                    }
                  );
              }
            });
        }
}

farfromrefug, 2019-08-02:
@dearpriyankasa great can i have to full error log with that?

dearpriyankasa, 2019-08-02:
JS: [nativescript-bluetooth] write ---- peripheralUUID:F4:50:EA:79:E0:2C serviceUUID:1800 characteristicUUID:2a00 
JS: Error: peripheral_not_connected
JS: Trace: Error: peripheral_not_connected
JS: at <anonymous> file:///src/app/characteristics-page/characteristics-page.component.ts:125:26
JS: at push.../node_modules/nativescript-angular/zone-js/dist/zone-nativescript.js.ZoneDelegate.invoke file:///node_modules/nativescript-angular/zone-js/dist/zone-nativescript.js:388:0
JS: at onInvoke file:///node_modules/@angular/core/fesm5/core.js:24349:0
JS: at push.../node_modules/nativescript-angular/zone-js/dist/zone-nativescript.js.ZoneDelegate.invoke file:///node_modules/nativescript-angular/zone-js/dist/zone-nativescript.js:387:0
JS: at push.../node_modules/nativescript-angular/zone-js/dist/zone-nativescript.js.Zone.run file:///node_modules/nativescript-angular/zone-js/dist/zone-nativescript.js:138:0
JS: at <anonymous> file:///node_modules/nativescript-angular/zone-js/dist/zone-nativescript.js:883:0
JS: at push.../node_modules/nativescript-angular/zone-js/dist/zone-nativescript.js.ZoneDelegate.invokeTask file:///node_modules/nativescript-angular/zone-js/dist/zone-nativescript.js:421:0
JS: at onInvokeTask file:///node_modules/@angular/core/fesm5/core.js:24340:0
JS: at push.../node_modules/nativescript-angular/zone-js/dist/zone-nativescript.js.ZoneDelegate.invokeTask file:///node_modules/nativescript-angular/zone-js/dist/zone-nativescript.js:420:0
JS: at push.../node_modules/nativescript-angular/zone-js/dist/zone-nativescript.js.Zone.runTask file:///node_modules/nativescript-angular/zone-js/dist/zone-nativescript.js:188:0

farfromrefug, 2019-08-02:
@dearpriyankasa you are on iOS or Android?

dearpriyankasa, 2019-08-02:
Android

farfromrefug, 2019-08-02:
@dearpriyankasa  ok so:
- enable debug using bluetooth.debug = true
- reproduce the issue
- send me the full logcat of the session with the device connction and the write issue

detonimarco, 2019-08-12:
Hi,
I have the same issue:
```
Read Battery of 00:81:F9:FD:64:D8 service  21961000-8ecb-9b83-284c-4ecee8d56fbb
vendor.js:82900 [nativescript-bluetooth] read: {
  "peripheralUUID": "00:81:F9:FD:64:D8",
  "serviceUUID": "21961000-8ecb-9b83-284c-4ecee8d56fbb",
  "characteristicUUID": "2a19"
}
bundle.js:345 ==== object dump start ====
originalStack: "Error: peripheral_not_connected
    at BluetoothError.ZoneAwareError (file:///data/data/org.nativescript.ngsample/files/app/vendor.js:82470:33)
    at new BluetoothError (file:///data/data/org.nativescript.ngsample/files/app/vendor.js:82882:28)
    at Bluetooth.push.../node_modules/nativescript-bluetooth/bluetooth.js.Bluetooth._getWrapper (file:///data/data/org.nativescript.ngsample/files/app/vendor.js:85335:35)
    at file:///data/data/org.nativescript.ngsample/files/app/vendor.js:82930:35
    at ZoneDelegate.push.../node_modules/nativescript-angular/zone-js/dist/zone-nativescript.js.ZoneDelegate.invoke (file:///data/data/org.nativescript.ngsample/files/app/vendor.js:81560:26)
    at Object.onInvoke (file:///data/data/org.nativescript.ngsample/files/app/vendor.js:62444:33)
    at ZoneDelegate.push.../node_modules/nativescript-angular/zone-js/dist/zone-nativescript.js.ZoneDelegate.invoke (file:///data/data/org.nativescript.ngsample/files/app/vendor.js:81559:32)
    at Zone.push.../node_modules/nativescript-angular/zone-js/dist/zone-nativescript.js.Zone.run (file:///data/data/org.nativescript.ngsample/files/app/vendor.js:81310:43)
    at file:///data/data/org.nativescript.ngsample/files/app/vendor.js:82055:34
    at ZoneDelegate.push.../node_modules/nativescript-angular/zone-js/dist/zone-nativescript.js.ZoneDelegate.invokeTask (file:///data/data/org.nativescript.ngsample/files/app/vendor.js:81593:31)
    at Object.onInvokeTask (file:///data/data/org.nativescript.ngsample/files/app/vendor.js:62435:33)
    at ZoneDelegate.push.../node_modules/nativescript-angular/zone-js/dist/zone-nativescript.js.ZoneDelegate.invokeTask (file:///data/data/org.nativescript.ngsample/files/app/vendor.js:81592:36)
    at Zone.push.../node_modules/nativescript-angular/zone-js/dist/zone-nativescript.js.Zone.runTask (file:///data/data/org.nativescript.ngsample/files/app/vendor.js:81360:47)
    at drainMicroTaskQueue (file:///data/data/org.nativescript.ngsample/files/app/vendor.js:81767:35)"
zoneAwareStack: "Error: peripheral_not_connected
    at BluetoothError.ZoneAwareError (file:///data/data/org.nativescript.ngsample/files/app/vendor.js:82470:33)
    at new BluetoothError (file:///data/data/org.nativescript.ngsample/files/app/vendor.js:82882:28)
    at Bluetooth.push.../node_modules/nativescript-bluetooth/bluetooth.js.Bluetooth._getWrapper (file:///data/data/org.nativescript.ngsample/files/app/vendor.js:85335:35)
    at file:///data/data/org.nativescript.ngsample/files/app/vendor.js:82930:35
    at ZoneDelegate.push.../node_modules/nativescript-angular/zone-js/dist/zone-nativescript.js.ZoneDelegate.invoke (file:///data/data/org.nativescript.ngsample/files/app/vendor.js:81560:26)
    at Object.onInvoke (file:///data/data/org.nativescript.ngsample/files/app/vendor.js:62444:33)
    at ZoneDelegate.push.../node_modules/nativescript-angular/zone-js/dist/zone-nativescript.js.ZoneDelegate.invoke (file:///data/data/org.nativescript.ngsample/files/app/vendor.js:81559:32)
    at Zone.push.../node_modules/nativescript-angular/zone-js/dist/zone-nativescript.js.Zone.run (file:///data/data/org.nativescript.ngsample/files/app/vendor.js:81310:43)
    at file:///data/data/org.nativescript.ngsample/files/app/vendor.js:82055:34
    at ZoneDelegate.push.../node_modules/nativescript-angular/zone-js/dist/zone-nativescript.js.ZoneDelegate.invokeTask (file:///data/data/org.nativescript.ngsample/files/app/vendor.js:81593:31)
    at Object.onInvokeTask (file:///data/data/org.nativescript.ngsample/files/app/vendor.js:62435:33)
    at ZoneDelegate.push.../node_modules/nativescript-angular/zone-js/dist/zone-nativescript.js.ZoneDelegate.invokeTask (file:///data/data/org.nativescript.ngsample/files/app/vendor.js:81592:36)
    at Zone.push.../node_modules/nativescript-angular/zone-js/dist/zone-nativescript.js.Zone.runTask (file:///data/data/org.nativescript.ngsample/files/app/vendor.js:81360:47)
    at drainMicroTaskQueue (file:///data/data/org.nativescript.ngsample/files/app/vendor.js:81767:35)"
stack: "Error: peripheral_not_connected
    at BluetoothError.ZoneAwareError (file:///data/data/org.nativescript.ngsample/files/app/vendor.js:82470:33)
    at new BluetoothError (file:///data/data/org.nativescript.ngsample/files/app/vendor.js:82882:28)
    at Bluetooth.push.../node_modules/nativescript-bluetooth/bluetooth.js.Bluetooth._getWrapper (file:///data/data/org.nativescript.ngsample/files/app/vendor.js:85335:35)
    at file:///data/data/org.nativescript.ngsample/files/app/vendor.js:82930:35
    at ZoneDelegate.push.../node_modules/nativescript-angular/zone-js/dist/zone-nativescript.js.ZoneDelegate.invoke (file:///data/data/org.nativescript.ngsample/files/app/vendor.js:81560:26)
    at Object.onInvoke (file:///data/data/org.nativescript.ngsample/files/app/vendor.js:62444:33)
    at ZoneDelegate.push.../node_modules/nativescript-angular/zone-js/dist/zone-nativescript.js.ZoneDelegate.invoke (file:///data/data/org.nativescript.ngsample/files/app/vendor.js:81559:32)
[... cut ...]
      "version": "6.0.1"
    }
  }
}

```

function called
```
    getBattery() {
        bluetooth.debug = true;
        console.log('Read Battery of ' + this.deviceUUID + ' service  ' + this.serviceUUID );
        bluetooth.read(
            {
                peripheralUUID: this.deviceUUID,
                serviceUUID: this.serviceUUID,
                characteristicUUID: '2A19'
            }).then(result => { console.dir(result); }, error => { console.dir(error); });

    }
}
```
OS: Android 9.1.0.311(C55E10R1P9)

detonimarco, 2019-08-12:
I found my issue.. using the workaround 
```
import { Bluetooth } from 'nativescript-bluetooth';
const bluetooth = new Bluetooth();
```
Bluetooth object is not global.. so when I connect after the scan and I open a new page using a component, in order to interact with the BLE device,  I instantiate a new Bluetooth object losing the previously created one.

dearpriyankasa, 2019-08-12:
U need to make the bluetooth object global. As I was using Angular in NS, I created an angular service where i created the ble object and instantiated that service in each component page. Thus it created only one copy of the ble object across all the components in my project.

farfromrefug, 2019-08-12:
@dearpriyankasa that s a big issue with the current layout of the code of that plugin. Especially the demo app creates a instance of Bluetooth on each page. This is a bad design for the exact reason you are explaining

closed 2020-08-27 by farfromrefug