nativescript-community/ble #151: startScanning discovering the same device multiple times on beta

AsgerCode, 2019-09-16:
Hello,

I have come across another issue. I wrote a function that runs at the start of my app that discovers BLE devices nearby as one would expect. The problem is that it used to work well, discovering all devices nearby once and listing them before the beta. On the beta, it discovers the same device over and over again. Here is my code: 

`AppModel.prototype.doStartScanning = function () {
        const that = this;
        isScanned = false;
        const omegaService = "d8b5d214-17c5-11e8-b642-0ed5f89f718b";

        that.set("isLoading", true);
        // reset the array
        observablePeripheralArray.splice(0, observablePeripheralArray.length);
        global.bluetooth.startScanning({
            serviceUUIDs: [],
            seconds: 3, // passing in seconds makes the plugin stop scanning after <seconds> seconds
            onDiscovered: function (peripheral) {

                if (peripheral.name !== null) {
                    console.log("A LEITURA E: " + peripheral.name);
                    observablePeripheralArray.push(observable.fromObject(peripheral));
                }

            }
        }).then(() => {
            isScanned = true;
            that.set("isLoading", false);
            if (observablePeripheralArray.length === 0) {

                const navigationEntry = {
                    moduleName: "pairing/pairing-failed",
                    animated: true
                };
                const topmost = frameModule.topmost();
                topmost.navigate(navigationEntry);

            }
        },
            (err) => {
                isScanned = true;
                that.set("isLoading", false);
                dialogs.alert({
                    title: "Please Enable Bluetooth",
                    message: err,
                    okButtonText: "OK"
                });

                const navigationEntry = {
                    moduleName: "pairing/pairing-page",
                    animated: true
                };
                const topmost = frameModule.topmost();
                topmost.navigate(navigationEntry);
            });

        return isScanned;
    };`

Is this a bug or am I misusing the beta again?

Cheers

farfromrefug, 2019-09-16:
@AsgerCode this is a wanted behavior to get RSI update. It s your app job in the beta to filter events and see if you already have the item
Also this is required to get advertisment data updates. You might not have it all on the first event

AsgerCode, 2019-09-16:
> @AsgerCode this is a wanted behavior to get RSI update. It s your app job in the beta to filter events and see if you already have the item
> Also this is required to get advertisment data updates. You might not have it all on the first event

Thanks for your help. What do you mean by the advertisement data updates? I want to make sure I'm not missing anything.

Cheers

farfromrefug, 2019-09-16:
When a device is discovered you get some advertismentData (see bluetooth docs). But when you get the discovered event, the advertismentData might be missing. If so it will come in a future discovered event (for the same device)

AsgerCode, 2019-09-16:
> When a device is discovered you get some advertismentData (see bluetooth docs). But when you get the discovered event, the advertismentData might be missing. If so it will come in a future discovered event (for the same device)

Thank you for all your help. Cheers

closed 2019-09-16 by AsgerCode