nativescript-community/ble #152: write function

AsgerCode, 2019-09-16:
The write function on beta only takes literals?

For example, passing an array as value representing 12000 in hexa as 2 byte values ["0x2e", "0xe0"] won't work, do they need to be literals?

If so how can I do this? There seems to be no way to create hexadecimal literals from a string.

farfromrefug, 2019-09-16:
@AsgerCode  yes   sending string hex is over;)

farfromrefug, 2019-09-16:
and by the way you should be able to pass 1200 directly I think

AsgerCode, 2019-09-16:
> and by the way you should be able to pass 1200 directly I think

The device I am working with strictly only accepts 2 byte hex so to send 4 bytes I have to make an array with 2 bytes + 2 bytes. Not being able to send hex strings is a problem here :cry:

farfromrefug, 2019-09-16:
@AsgerCode no no it is not. You can send [0x2e, 0xe0]

AsgerCode, 2019-09-16:
> @AsgerCode no no it is not. You can send [0x2e, 0xe0]

Yeah but I am not receiving an exact number, it is a variable that the user inputs on a slider. This means that as I am being sent the number, I have to convert it to base16, after that it is a number left without a 0x before it, so I make it a string and concatenate "0x". This leaves me with a "0x" hex but as a string.

farfromrefug, 2019-09-17:
@AsgerCode But you can send [46,224]. It is exactly the same
No need for any addition of 0x as a string

closed 2020-08-27 by farfromrefug