nativescript-community/ble #154: ManufacturerData in release 2.0.0-beta.23

j99ht, 2019-10-03:
I have migrated my app to Nativescript 6 and nativescript-bluetooth 2.0.0-beta.23

I have two issues with thie release:
- Now I receive the same device multiple times in the same scan period, while it was received only one in 1.3.1 Is there any way to configure this?
- I cannot access to manufacturerDataBytes anymore. I see ManufacturerID, but I do not see the rest of the information. How could I get this?

farfromrefug, 2019-10-21:
@j99ht multiple events are intended. It is especially needed when advertisment data change.
```manufacturerDataBytes``` is now ```manufacturerData```. It is part of the advertisment data. Now it is an ```ArrayBuffer```

j99ht, 2019-11-14:
Thanks. I works now

closed 2019-11-14 by j99ht