nativescript-community/ble #159: Ios State Preservation and Restoration

dimitriospafos, 2019-11-11:
Is there a way to setup State Preservation and Restoration for ios upon initializing CBCentralManager ?

farfromrefug, 2019-11-13:
@dimitriospafos it is possible in v2 https://github.com/EddyVerbruggen/nativescript-bluetooth/blob/v2/src/bluetooth.ios.ts#L626. Use npm beta tagged version

dimitriospafos, 2019-11-13:
@farfromrefug thanks a lot, will give it a try.

dimitriospafos, 2019-11-15:
@farfromrefug Is there anything else I need to do in my code except passing a restore identifier to the constructor of "Bluetooth"? I have something like:
             
public bluetoothManager: Bluetooth;      
this.bluetoothManager = new Bluetooth("BLE_Restore_Identifier");

mbaush77, 2020-11-11:
Is it possible to run a bluetooth scan in background with a worker or CustomDelegate on IOS?

Using a worker, I'm getting the "bluetooth_not_enabled" error as soon as my app goes to background, but works fine in foreground.

dimitriospafos, 2020-11-16:
@mbaush77 you can have a background scan as long as you are trying to connect to a ble peripheral by passing the serviceUUID of the peripheral when start to scan, it should connect even if its in background. 

If the app is killed by the user then the scan will terminate. If the app is suspended by the OS due to memory usage then you will have to implement "State Preservation and Restoration" since the app goes to suspended state, see my question above.

Make sure you specify the correct background modes either in Info.plist or in Xcode directly.