nativescript-community/ble #160: startScanning onDiscovered callback never triggered

remusmp, 2019-11-14:
Hello everyone,

I'm developing an app on a tablet (Lenovo Tab M10). I need to scan bluetooth devices and connect to some of them but the nativescript-bluetooth library doesn't see any devices. However the builtin bluetooth scanner of the tablet sees pretty much all devices that I want to connect to.

I have the below bluetooth.service.ts (just something simple for demo purposes):
```
import { Injectable } from "@angular/core";
import { Bluetooth, Peripheral } from "nativescript-bluetooth";

const bluetooth = new Bluetooth();
bluetooth.debug = true;

@Injectable({
    providedIn: "root"
})
export class BluetoothService {
    constructor() {
        console.log("Creating BluetoothService");
    }

    startScanning() {
        const filters = [];
        const seconds = 4;
        bluetooth.startScanning({
            filters,
            seconds,
            onDiscovered:
                (peripheral: Peripheral) => {
                    console.log(peripheral);
                },
            skipPermissionCheck: false
        }).then(
            (res) => {
                console.log("bluetooth scanning complete");
            },
            (err) => {
                console.error("Problem with bluetooth startScanning", err, true);
            }
        );
    }
}
```

I call startScanning from a component but I see no peripherals being logged. Here is console.log:
```
LOG from device Lenovo Tab M10: [nativescript-bluetooth] coarseLocationPermissionGranted ---- ACCESS_COARSE_LOCATION permission granted? true
LOG from device Lenovo Tab M10: [nativescript-bluetooth] startScanning ---- PostLollipop ---- didStart scanning: []
LOG from device Lenovo Tab M10: [nativescript-bluetooth] stopCurrentScan: true
LOG from device Lenovo Tab M10: bluetooth scanning complete
```

Am I missing something in setting up the library?

remusmp, 2019-11-14:
I'll answer my own question :)
I missed the following line from readme:
```
Moreover on Android if we don't use filters we must have location permissions and have GPS enabled.
```

closed 2019-11-14 by remusmp

Wassim24, 2019-11-26:
@remusmp how did you solve the problem ? I'm facing the exact same issue.

remusmp, 2019-11-26:
> @remusmp how did you solve the problem ? I'm facing the exact same issue.

Hello! I turned on the GPS on my tablet and make sure you have location permissions for the app. It worked fine afterwards.

Wassim24, 2019-11-29:
@remusmp  Thank you very much !
But for my situation, this was not the problem, as the gps was enable.
I tested with another android device and it worked right away :)