nativescript-community/ble #162: After restart: Cannot read property 'localName' of undefined

mitchdries, 2019-11-16:
Hi, first of all thanks for this great plugin!

I’m using 2.0.0-beta and I ran into the following issue:

1. `startScanning`
2. `connect` to device
3. now everything works fine
4. kill the application and start it again
5. the device UUID is known in the app, so `connect` again
6. the following error occurs:
```
JS: TypeError: Cannot read property 'localName' of undefined
JS:     at file:///node_modules/nativescript-bluetooth/bluetooth.js:1547:0
```

It seems to be that `advertismentData` is undefined at this point. I wasn't able to figure out why exactly this is happening. 

For now[ I was thinking to solve it this way](https://github.com/mitchdries/nativescript-bluetooth/commit/cadad3ce3db8997b419dabe9f73b1808ba036f89), but it might be just a workaround for an underlying problem…

Any ideas?

Thanks.

----------
package.json:
```
{
  "nativescript": {
    "tns-android": {
      "version": "6.2.0"
    },
    "tns-ios": {
      "version": "6.2.0"
    }
  },
  "dependencies": {
    "@angular/animations": "~8.2.0",
    "@angular/common": "~8.2.0",
    "@angular/compiler": "~8.2.0",
    "@angular/core": "~8.2.0",
    "@angular/forms": "~8.2.0",
    "@angular/platform-browser": "~8.2.0",
    "@angular/platform-browser-dynamic": "~8.2.0",
    "@angular/router": "~8.2.0",
    "nativescript-angular": "~8.2.0",
    "nativescript-bluetooth": "^2.0.0-beta.27",
    "nativescript-theme-core": "~1.0.6",
    "reflect-metadata": "~0.1.12",
    "rxjs": "^6.4.0",
    "tns-core-modules": "~6.2.0",
    "zone.js": "~0.9.1"
  },
  "devDependencies": {
    "@angular/cli": "^8.3.8",
    "@angular/compiler-cli": "~8.2.0",
    "@nativescript/schematics": "^0.7.2",
    "@ngtools/webpack": "~8.2.0",
    "nativescript-dev-webpack": "~1.3.0",
    "typescript": "~3.5.3"
  }
}
```

farfromrefug, 2019-11-17:
@mitchdries ok found the reason of the issue and why i am not seeing it. 
So you understand, myself when i want to do the same thing as you (reconnect upon app restart) i always go through a scan, then if i find my saved device i connect. I do that because i want updated advertisment data which i can only get through a scan.

Now what you do is not wrong and there is a bug in the connect code which assume there is an advertisment data! I will fix it right away. Though in your case you wont get any adv data from the device you connect to upon restart.

farfromrefug, 2019-11-17:
@mitchdries released a new beta version 3.0.0-beta.7 which fixes your issue.
But be careful 3.0.0 is made for new ```@nativescript``` packages. It wont work with ```tns-core-modules``` (not without a change in your webpack config)

mitchdries, 2019-11-17:
Thanks for the quick response, works like a charm now!

In my case the 2nd time scanning did not detect the device anymore, that's why I was trying to connect to it directly without scanning before, could be perhaps device related. The adv data is not important for my use case.

Thanks a lot, again!

closed 2019-11-17 by mitchdries