nativescript-community/ble #163: nativescript-bluetooth@3.0.0-beta.7 (and others) doesn't work with Playground

MpeStuntPilot, 2019-11-25:
Latest changes to Playground has broken compatibility with nativescript-bluetooth.
Preview App point to incompatibility with nativescript-observable in crash log.
Target device is iPhone Xr with latest iOS ,
PC = Mac Book Pro with latest OS,
Browser = Firefox

Test App = generic / initial Playground test app with only "require ... nativescript-bluetooth" as shown below:
Commenting out the require .... nativescript-bluetooth line allows the app to function,
Including the require line causes it to immediately crash.

///////////////////
const frame = require("ui/frame");
const fromObject = require("tns-core-modules/data/observable").fromObject;
var bluetooth = require("~/nativescript-bluetooth@3.0.0-beta.7"); // import BLE nativescript library
var page;

function pageLoaded(args) {
  page = args.object;
}
exports.pageLoaded = pageLoaded;
//////////////////

farfromrefug, 2019-11-25:
How does the preview app point to nativescript-observable. Please when you post an issue post logs and {N}, runtime and plug-ins versions
And please follow markdown style

MpeStuntPilot, 2019-12-01:
My system is a mac book pro 15" with iOS 10.14.4, with FIrefox 70.0.1 as the browser.
iPhone Xr with latest iOS 13.2.3, latest {N} apps Playground 1.8.0, Preview 1.27.0 (from App Store Ver History)
nativescript-bluetooth@3.0.0-beta.7
NSPlayground(3).zip with the source files is attached to this comment

**How does the preview app point to nativescript-observable.** 
A:  The illegal reference only occurs when the line 
var bluetooth = require("~/nativescript-bluetooth@3.0.0-beta.7");
is not commented out.  Otherwise, the app is fine.

Steps to reproduce the error
1. Start the playground web app
[https://play.nativescript.org/?_ga=2.14993334.1088328485.1575239630-338449013.1561430891](url)
2. Click the Explorer + (Add new item) -> Add NPM package
3. Type nativescript-bluetooth into the search box, and selected nativescript-bluetooth@3.0.0-beta.7
4. Add the following line to the home-page.js file
var bluetooth = require("~/nativescript-bluetooth@3.0.0-beta.7");
5. Press Preview, Scan QR code with {N} Playground
6.  App will immediately crash
7. Restart Preview App
8. View crash log (shown below)
2019-12-01 14:45:01.170 nsplaydev[3388:1513024] PlayLiveSync: applicationDidFinishLaunching
2019-12-01 14:45:01.258 nsplaydev[3388:1513024] ***** Fatal JavaScript exception - application has been terminated. *****
2019-12-01 14:45:01.258 nsplaydev[3388:1513024] Native stack trace:
1   0x1017ae2bc NativeScript::reportFatalErrorBeforeShutdown(JSC::ExecState*, JSC::Exception*, bool)
2   0x1017e4578 NativeScript::FFICallback<NativeScript::ObjCMethodCallback>::ffiClosureCallback(ffi_cif*, void*, void**, void*)
3   0x10229cea4 ffi_closure_SYSV_inner
4   0x1022a01b4 .Ldo_closure
5   0x1a2fb207c <redacted>
6   0x1a2b3fae0 <redacted>
7   0x1a2b3fb30 <redacted>
8   0x1a2b3ee28 <redacted>
9   0x1a2b3eac0 <redacted>
10  0x1a2ab7a58 <redacted>
11  0x1a2b3e3f0 _CFXNotificationPost
12  0x1a2ea0c1c <redacted>
13  0x1a6cc536c <redacted>
14  0x1a6ccb42c <redacted>
15  0x1a646555c <redacted>
16  0x1a6915db0 _UIScenePerformActionsWithLifecycleActionMask
17  0x1a6466094 <redacted>
18  0x1a6465ac4 <redacted>
19  0x1a6465eb0 <redacted>
20  0x1a6465734 <redacted>
21  0x1a6469ee4 <redacted>
22  0x1a692fc34 <redacted>
23  0x1a6842eec <redacted>
24  0x1a692fbec _UISceneSettingsDiffActionPerformChangesWithTransitionContext
25  0x1a6469bfc <redacted>
26  0x1a692fad4 _UISceneSettingsDiffActionPerformActionsWithDelayForTransitionContext
27  0x1a6469a58 <redacted>
28  0x1a62d1b7c <redacted>
29  0x1a62d0640 <redacted>
30  0x1a62d18ac <redacted>
31  0x1a6cc97e0 <redacted>
2019-12-01 14:45:01.262 nsplaydev[3388:1513024] JavaScript stack trace:
2019-12-01 14:45:01.262 nsplaydev[3388:1513024] file:///app/tns_modules/@nativescript/core/ui/builder/builder.js:202:46
at file:///app/tns_modules/@nativescript/core/xml/xml.js:146:14
at parse(file:///app/tns_modules/@nativescript/core/js-libs/easysax/easysax.js:751:28)
at parse(file:///app/tns_modules/@nativescript/core/xml/xml.js:193:23)
at parse(file:///app/tns_modules/@nativescript/core/ui/builder/builder.js:208:24)
at parseInternal(file:///app/tns_modules/@nativescript/core/ui/builder/builder.js:170:14)
at loadInternal(file:///app/tns_modules/@nativescript/core/ui/builder/builder.js:97:36)
at createViewFromEntry(file:///app/tns_modules/@nativescript/core/ui/builder/builder.js:37:43)
at createRootView(file:///app/tns_modules/@nativescript/core/application/application.js:302:57)
at setWindowContent(file:///app/tns_modules/@nativescript/core/application/application.js:244:34)
at notifyAppStarted(file:///app/tns_modules/@nativescript/core/application/application.js:175:28)
at didFinishLaunchingWithOptions(file:///app/tns_modules/@nativescript/core/application/application.js:164:26)
at [native code]
at onReceive(file:///app/tns_modules/@nativescript/core/application/application.js:37:28)
[... 60 lines cut ...]
', reason: '(null)'
*** First throw call stack:
(0x1a2be7ab0 0x1a2901028 0x1017ae738 0x1017e4578 0x10229cea4 0x1022a01b4 0x1a2fb207c 0x1a2b3fae0 0x1a2b3fb30 0x1a2b3ee28 0x1a2b3eac0 0x1a2ab7a58 0x1a2b3e3f0 0x1a2ea0c1c 0x1a6cc536c 0x1a6ccb42c 0x1a646555c 0x1a6915db0 0x1a6466094 0x1a6465ac4 0x1a6465eb0 0x1a6465734 0x1a6469ee4 0x1a692fc34 0x1a6842eec 0x1a692fbec 0x1a6469bfc 0x1a692fad4 0x1a6469a58 0x1a62d1b7c 0x1a62d0640 0x1a62d18ac 0x1a6cc97e0 0x1a6864dec 0x1a7db1ec0 0x1a7dd8b50 0x1a7dbcfa4 0x1a7dd87e4 0x1a288dfd8 0x1a2890d1c 0x1a7dff304 0x1a7dfefb0 0x1a7dff51c 0x1a2b6324c 0x1a2b631a0 0x1a2b6295c 0x1a2b5d7d8 0x1a2b5d084 0x1acdab534 0x1a6ccd670 0x1022a0044 0x10229c5c8 0x10229c0c4 0x101770eb0 0x10229b2c8 0x102299644 0x102299644 0x102299644 0x1022850ec 0x101f2014c 0x1020ee7a8 0x1017894c0 0x1018ed728 0x10229b02c 0x1022995a4 0x102299644 0x1022850ec 0x101f2014c 0x1020ee974 0x1018e6ec4 0x10177feb0 0x101785a30 0x10229b02c 0x102299644 0x1022850ec 0x101f2014c 0x1020ee7a8 0x1017894c0 0x1018ed728 0x10229b02c 0x102299644 0x102299644 0x1022850ec 0x101f2014c 0x1020ee974 0x1018e6ec4 0x10177feb0 0x1017f282c 0x10280625c 0x100de7524 0x1a29dce18)
libc++abi.dylib: terminating with uncaught exception of type NSException
2019-12-01 14:45:01.275 nsplaydev[3388:1513024] PlayLiveSync: Uncaught Exception

This happens 100% of the time on my system.

commenting out the bluetooth line allows the app to work.

ZIP file with sources
[NSPlayground(3).zip](https://github.com/EddyVerbruggen/nativescript-bluetooth/files/3908829/NSPlayground.3.zip)

home-page.js (shown below)

//var bluetooth = require("~/nativescript-bluetooth@3.0.0-beta.7");
var HomeViewModel = require("./home-view-model");
var homeViewModel = new HomeViewModel();

function pageLoaded(args) {
  var page = args.object;
  page.bindingContext = homeViewModel;
}

exports.pageLoaded = pageLoaded;

farfromrefug, 2019-12-02:
it says ```Could not find module 'nativescript-observable``` For some reason the dependency is not found. Though it is marked as a dep in ```nativescript-bluetooth```. This seems to be an issue in your setup.

MpeStuntPilot, 2019-12-03:
I have tried this on a PC running Win10 with Firefox, but using the same iPhone.
The exact same thing happens, and the setup is completely different.
The only common things are the iphone, the playground app, and the preview app.
No login was used on the PC, I used "guest" mode so that playground would not be linked in some way with me.
Did you try the example?  Did you not have the same experience?
I also had the same thing happen with a Samsung A10.
This seems to strongly point to the nativescript-bluetooth library, as the problem goes away when the its line is commented out.

fangjin2000, 2019-12-03:
I got the same problem. After adding the nativescript-observable module, here are different errors below:

ERROR: ERROR Error: Uncaught (in promise): SyntaxError: Importing binding name 'Observable' is not found.
moduleDeclarationInstantiation([native code])
at link([native code])
at linkAndEvaluateModule([native code])
at promiseReactionJob([native code])
at require([native code])
at anonymous(file:///app/nativescript-bluetooth@3.0.0-beta.7/bluetooth.common.js:2:40)
at evaluate([native code])
at moduleEvaluation([native code])
at promiseReactionJob([native code])
at require([native code])
at anonymous(file:///app/nativescript-bluetooth@3.0.0-beta.7/bluetooth.js:2:33)
at evaluate([native code])
at moduleEvaluation([native code])
at promiseReactionJob([native code])
at require([native code])
at anonymous(file:///app/home/home.component.js:4:54)
at evaluate([native code])
at moduleEvaluation([native code])
at promiseReactionJob([native code])
at require([native code])
at anonymous(file:///app/home/home-routing.module.js:5:31)
at evaluate([native code])
at moduleEvaluation([native code])
at promiseReactionJob([native code])
at require([native code])
at anonymous(file:///app/home/home.module.js:13:36)
at evaluate([native code])
at moduleEvaluation([native code])
at promiseReactionJob([native code])
at require([native code])
at onInvoke(file:///app/tns_modules/@angular/core/bundles/core.umd.js:26371:33)
at run(file:///app/tns_modules/@nativescript/angular/zone-js/dist/zone-nativescript.js:138:43)
at file:///app/tns_modules/@nativescript/angular/zone-js/dist/zone-nativescript.js:883:31
at onInvokeTask(file:///app/tns_modules/@angular/core/bundles/core.umd.js:26362:37)
at runTask(file:///app/tns_modules/@nativescript/angular/zone-js/dist/zone-nativescript.js:188:49)
at drainMicroTaskQueue(file:///app/tns_modules/@nativescript/angular/zone-js/dist/zone-nativescript.js:595:32)
at file:///app/tns_modules/@nativescript/angular/platform-common.js:179:38
at notify(file:///app/tns_modules/@nativescript/core/data/observable/observable.js:110:23)
at [native code]
at notifyAppStarted(file:///app/tns_modules/@nativescript/core/application/application.js:172:32)
at didFinishLaunchingWithOptions(file:///app/tns_modules/@nativescript/core/application/application.js:164:26)
at [native code]
at onReceive(file:///app/tns_modules/@nativescript/core/application/application.js:37:28)
at UIApplicationMain([native code])
at _start(file:///app/tns_modules/@nativescript/core/application/application.js:324:22)
at run(file:///app/tns_modules/@nativescript/core/application/application.js:366:9)
at bootstrapNativeScriptApp(file:///app/tns_modules/@nativescript/angular/platform-common.js:210:22)
at bootstrapApp(file:///app/tns_modules/@nativescript/angular/platform-common.js:111:34)
at bootstrapModule(file:///app/tns_modules/@nativescript/angular/platform-common.js:91:22)
at anonymous(file:///app/main.js:8:57)
at evaluate([native code])
at moduleEvaluation([native code])
at promiseReactionJob([native code])
at require([native code])
at anonymous(file:///app/__main.js:3:8)
at evaluate([native code])
at moduleEvaluation([native code])
at promiseReactionJob([native code])

closed 2020-08-27 by farfromrefug