nativescript-community/ble #165: bluetooth methods undefined

DimitrisRK, 2019-12-06:
After installing plugin, I tried to use it on my android app.
require() loads plugin with success but when I try to call any of the documented functions, an error is thrown indicating that called function is undefined.

I logged bluetooth object in JSON format and result was the following:
`{"ScanMode":{"0":"LOW_LATENCY","1":"BALANCED","2":"LOW_POWER","3":"OPPORTUNISTIC","LOW_LATENCY":0,"BALANCED":1,"LOW_POWER":2,"OPPORTUNISTIC":3},"MatchMode":{"0":"AGGRESSIVE","1":"STICKY","AGGRESSIVE":0,"STICKY":1},"MatchNum":{"0":"MAX_ADVERTISEMENT","1":"FEW_ADVERTISEMENT","2":"ONE_ADVERTISEMENT","MAX_ADVERTISEMENT":0,"FEW_ADVERTISEMENT":1,"ONE_ADVERTISEMENT":2},"CallbackType":{"0":"ALL_MATCHES","1":"FIRST_MATCH","2":"MATCH_LOST","ALL_MATCHES":0,"FIRST_MATCH":1,"MATCH_LOST":2},"Phy":{"0":"LE_1M","1":"LE_CODED","2":"LE_ALL_SUPPORTED","LE_1M":0,"LE_CODED":1,"LE_ALL_SUPPORTED":2}}`

As I can see, functions are nowhere to be found.

DimitrisRK, 2019-12-16:
Problem still persists.

DimitrisRK, 2020-04-23:
I think the problem is that your plugin is not converted from TypeScript to JavaScript using pack.sh.

manojdcoder, 2020-04-23:
Do you still have the issue? Can you share the sample code? Which version of plugin you are using?

DimitrisRK, 2020-04-23:
> Do you still have the issue? Can you share the sample code? Which version of plugin you are using?

I'm using the latest version of plugin from npmjs. Seems like I was wrong and JS is generated, but issue still persists. All methods appear to be undefined even though I import plugin.
I'll try to find more information and post it here. By the way, I'm using plain JS.

DimitrisRK, 2020-04-23:
So, I attempted to create an instance of Bluetooth using this code sample:
```
// require the plugin
var bluetooth = require("nativescript-bluetooth");
console.log(new bluetooth.Bluetooth());
```
And I got this error:
```
System.err: com.tns.NativeScriptException: Calling js method onCreateView failed
System.err: Error: java.lang.ClassNotFoundException: Didn't find class "com.tns.gen.android.bluetooth.BluetoothAdapter$LeScanCallback" on path: DexPathList[[zip file "/data/data/org.fuelics.tankowner/code_cache/secondary-dexes/android.bluetooth.BluetoothAdapter_LeScanCallback-1587678141169-5.jar"],nativeLibraryDirectories=[/vendor/lib, /system/lib]]
System.err:     at com.tns.Runtime.callJSMethodNative(Native Method)
System.err:     at com.tns.Runtime.dispatchCallJSMethodNative(Runtime.java:1286)
System.err:     at com.tns.Runtime.callJSMethodImpl(Runtime.java:1173)
System.err:     at com.tns.Runtime.callJSMethod(Runtime.java:1160)
System.err:     at com.tns.Runtime.callJSMethod(Runtime.java:1138)
System.err:     at com.tns.Runtime.callJSMethod(Runtime.java:1134)
System.err:     at com.tns.FragmentClass.onCreateView(FragmentClass.java:53)
System.err:     at androidx.fragment.app.Fragment.performCreateView(Fragment.java:2439)
System.err:     at androidx.fragment.app.FragmentManagerImpl.moveToState(FragmentManager.java:1460)
System.err:     at androidx.fragment.app.FragmentManagerImpl.moveFragmentToExpectedState(FragmentManager.java:1784)
System.err:     at androidx.fragment.app.FragmentManagerImpl.moveToState(FragmentManager.java:1852)
System.err:     at androidx.fragment.app.BackStackRecord.executeOps(BackStackRecord.java:802)
System.err:     at androidx.fragment.app.FragmentManagerImpl.executeOps(FragmentManager.java:2625)
System.err:     at androidx.fragment.app.FragmentManagerImpl.executeOpsTogether(FragmentManager.java:2411)
System.err:     at androidx.fragment.app.FragmentManagerImpl.removeRedundantOperationsAndExecute(FragmentManager.java:2366)
System.err:     at androidx.fragment.app.FragmentManagerImpl.execPendingActions(FragmentManager.java:2273)
System.err:     at androidx.fragment.app.FragmentManagerImpl$1.run(FragmentManager.java:733)
System.err:     at android.os.Handler.handleCallback(Handler.java:808)
System.err:     at android.os.Handler.dispatchMessage(Handler.java:103)
System.err:     at android.os.Looper.loop(Looper.java:193)
System.err:     at android.app.ActivityThread.main(ActivityThread.java:5322)
System.err:     at java.lang.reflect.Method.invokeNative(Native Method)
System.err:     at java.lang.reflect.Method.invoke(Method.java:515)
System.err:     at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:829)
System.err:     at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:645)
System.err:     at dalvik.system.NativeStart.main(Native Method)
```

manojdcoder, 2020-04-25:
[BluetoothAdapter.LeScanCallback](https://developer.android.com/reference/android/bluetooth/BluetoothAdapter.LeScanCallback) is available since Jelly Bean, so I'm not sure how KitKat could hit class not found exception.

Which device you are using?

DimitrisRK, 2020-04-25:
I'm using ZTE Blade A450. Few months ago, I copied android source code from here to make my app work (of course I wouldn't wait for 4 months to get an answer ^^), and everything worked just fine using native API. `android.bluetooth.BluetoothAdapter.LeScanCallback` works fine, but regarding `com.tns.gen.android.bluetooth`...never have I seen that package before before.

farfromrefug, 2020-08-27:
@DimitrisRK did you get it to work?

DimitrisRK, 2020-08-28:
@farfromrefug Unfortunately, the error I posted in my previous posts still persists.
I have 2 devices, one with Android 4.4 (Kitkat) and another with Android 9.0.
On the latter, plugin works as expected so the problem persists on that really old device.
It's not big deal if it's not running on an old android os, but it's weird since plugin looks pretty well-developed.

Few months ago, I developed my custom BLE app using pure native android classes and had no issues.
So I wonder if issue is related to a certain code block in ble.aar or nativescript_bluetooth.aar compiled bundles.

farfromrefug, 2020-08-28:
@DimitrisRK so only fails on 4.4 with this? 
```
System.err: Error: java.lang.ClassNotFoundException: Didn't find class "com.tns.gen.android.bluetooth.BluetoothAdapter$LeScanCallback" on path: DexPathList[[zip file "/data/data/org.test.test1/code_cache/secondary-dexes/android.bluetooth.BluetoothAdapter_LeScanCallback-1587678141169-5.jar"],nativeLibraryDirectories=[/vendor/lib, /system/lib]]
```

I can try and fix that.

DimitrisRK, 2020-08-28:
> @DimitrisRK so only fails on 4.4 with this?
> 
> ```
> System.err: Error: java.lang.ClassNotFoundException: Didn't find class "com.tns.gen.android.bluetooth.BluetoothAdapter$LeScanCallback" on path: DexPathList[[zip file "/data/data/org.test.test1/code_cache/secondary-dexes/android.bluetooth.BluetoothAdapter_LeScanCallback-1587678141169-5.jar"],nativeLibraryDirectories=[/vendor/lib, /system/lib]]
> ```
> 
> I can try and fix that.
> 
> EDIT: i think the issue is with native android class not existing on 4.4

That's right. This occurs on 4.4.
On 8.0 and 9.0, plugin has no errors.

DimitrisRK, 2020-09-09:
> EDIT: i think the issue is with native android class not existing on 4.4

@farfromrefug Few months ago, I made ble scanner work without the use of this plugin, but by using purely native android classes.
If this helps, I can upload a test sample when I get more time.

farfromrefug, 2020-09-09:
@DimitrisRK sure do share it might help

closed 2020-09-15 by DimitrisRK