nativescript-community/ble #167: Build error with Nativescript 6 and 2.0.0.beta27

jpmartinsson, 2019-12-10:
Environment:
```
    "@angular/animations": "~8.2.0",
    "@angular/common": "~8.2.0",
    "@angular/compiler": "~8.2.0",
    "@angular/core": "~8.2.0",
    "@angular/forms": "~8.2.0",
    "@angular/http": "~8.0.0-beta.10",
    "@angular/platform-browser": "~8.2.0",
    "@angular/platform-browser-dynamic": "~8.2.0",
    "@angular/router": "~8.2.0",
    "@proplugins/nativescript-localstorage": "^2.3.1",
    "core-js": "^2.4.1",
    "css-loader": "^3.2.0",
    "nativescript-angular": "^8.20.1",
    "nativescript-bluetooth": "^2.0.0-beta.27",
    "nativescript-datetimepicker": "^1.2.2",
    "nativescript-geolocation": "^5.1.0",
    "nativescript-ngx-shadow": "^6.5.0",
    "nativescript-theme-core": "^1.0.6",
    "nativescript-ui-gauge": "^5.0.0",
    "reflect-metadata": "~0.1.12",
    "rxjs": "^6.4.0",
    "rxjs-compat": "^6.0.0-rc.0",
    "tns-core-modules": "~6.2.0",
    "tslib": "^1.10.0",
    "zone.js": "^0.9.1"
```

After upgrade from 1.3.1 to 2.0.0.beta27 and I now receive this error upon building the code (tns build ios, tns build android:
```
ERROR in node_modules/nativescript-observable/observable.d.ts:7:41 - error TS2689: Cannot extend an interface 'NObservable'. Did you mean 'implements'?

7 export default class Observable extends NObservable {
```
I am using it like this in a service:
```
import { Bluetooth } from 'nativescript-bluetooth';
...
const bluetooth = new Bluetooth(); 
```

and in a component:
```
import { Bluetooth, Peripheral, Service, ReadResult } from 'nativescript-bluetooth';
...
bluetooth = new Bluetooth();
```

farfromrefug, 2019-12-10:
It should not fail there. Can you post the full log? Seems like a webpack issue

jpmartinsson, 2019-12-10:
Ok. Here it is.
[Issue_167.txt](https://github.com/EddyVerbruggen/nativescript-bluetooth/files/3947595/Issue_167.txt)

jpmartinsson, 2019-12-11:
The only thing I did before getting this message was upgrading to version 2.0.0 of nativescript-bluetooth and migrating the code shown above. I have, after this, tried things like dropping `platform`, `hooks`, `node_modules` and `package-lock.json` and then doing a new `npm install`. The same problem remains and no visible changes to the error log can be seen.

jpmartinsson, 2019-12-11:
When backing my environment to native-bluetooth 1.3.1, the error below disappears, so it is definitely (somehow) related to the upgrade to 2.0.0. Am I missing something in the migration?
```
ERROR in node_modules/nativescript-observable/observable.d.ts:7:41 - error TS2689: Cannot extend an interface 'NObservable'. Did you mean 'implements'?

7 export default class Observable extends NObservable {
```
Since 1.3.1 does not work with Android on Nativescript 6 (correct?), we'd really need to use the 2.0.0 version though. On Android 1.3.1 - We receive the same error as referenced in [issue #145 ](https://github.com/EddyVerbruggen/nativescript-bluetooth/issues/145), i.e. `TypeError: Cannot read property \'ContextCompat\' of undefined'`

farfromrefug, 2019-12-11:
@jpmartinsson ok i think i found out the issue. Switch to beta 3.0.0 and it should be fine (it s related to the new @nativescript package)

jpmartinsson, 2019-12-11:
Ok. Where can I find 3.0.0? Are there any readme notes for 3.0.0? Major differences from 2.0.0?

farfromrefug, 2019-12-11:
https://www.npmjs.com/package/nativescript-bluetooth
look in versions.
No changelog. One thing though is that it require ```@nativescript/core``` which you have with ```"tns-core-modules": "~6.2.0"```

jpmartinsson, 2019-12-11:
Ok. I upgraded to 3.0.0.beta-7. The previous error was gone but replaced with a new error (when doing tns run ios --bundle):
```
ERROR in node_modules/nativescript-bluetooth/bluetooth.common.d.ts:24:9 - error TS1086: An accessor cannot be declared in an ambient context.

24     set debug(value: boolean);
```

any clues regarding this?

jpmartinsson, 2019-12-11:
Commenting out the line of code in the above error message (in `bluetooth.common.d.ts`) makes the compile go through without errors. Am I creating this error by using the plugin the wrong way or is this a bug in the code?

farfromrefug, 2019-12-11:
@jpmartinsson sorry about the new error it s weird because I don't get this one. What is your version of typescript?can you share your tsconfig?

jpmartinsson, 2019-12-11:
Typescript: `"typescript": "~3.5.3"`

tsconfig:
```
{
  "compileOnSave": false,
  "compilerOptions": {
    "downlevelIteration": true,
    "importHelpers": true,
    "outDir": "./dist/out-tsc",
    "sourceMap": true,
    "declaration": false,
    "types": ["node"],
    "module": "commonjs",
    "moduleResolution": "node",
    "emitDecoratorMetadata": true,
    "experimentalDecorators": true,
    "target": "es2015",
    "typeRoots": [
      "node_modules/@types"
    ],
    "lib": [
      "es2017",
      "dom"
    ],
    "baseUrl": "./",
    "paths": {
      "@src/*": [
        "src/*.android.ts",
        "src/*.ios.ts",
        "src/*.tns.ts",
        "src/*.web.ts",
        "src/*"
      ]
    }
  }
}
```
tsconfig.tns.ts:
```
  {
  "extends": "./tsconfig",
  "compilerOptions": {
    "module": "es2015",
    "moduleResolution": "node",
    "paths": {
      "@src/*": [
        "src/*.tns.ts",
        "src/*.ts"
      ]
    }
  },
  "files": [
    "src/main.tns.ts"
  ]
}
```

jpmartinsson, 2019-12-13:
Any success finding the reason for this last error? Are there any important side effects to just commenting the line of code out as I have done now?

farfromrefug, 2019-12-13:
yes just found it
https://stackoverflow.com/questions/58802463/types-sequelize-error-ts1086-an-accessor-cannot-be-declared-in-ambient-context

It works here because i use ts 3.7
Can you switch?
If not i could make a version without it though i like it:P

jpmartinsson, 2019-12-13:
Hmm, I am not too keen on upgrading to 3.7 now since we're too close to a release...

farfromrefug, 2019-12-13:
ok then modify the definition file directly for now. Will think about it later

jpmartinsson, 2019-12-13:
Ok. So commenting the row out will not have any important side effects you know of?

farfromrefug, 2019-12-13:
not if you dont use it.

closed 2020-08-27 by farfromrefug