nativescript-community/ble #169: Bluetooth plugin not working on iOS device

prabudevarrajan, 2020-01-14:
I cloned the project and tried to run the demo app. On Android, the demo app works fine without any issue. But same app on my iOS device doesn't work.

When I clicked on "Is Bluetooth Enabled?", is says "Bluetooth is DISABLED". But, I checked the phone settings and bluetooth service is enabled. 

iOS device information
iPhone 6s 
iOS version 13.2.3

manojdcoder, 2020-02-24:
@prabudevarrajan Do you still have the issue, can you share a sample project?

luca-mayven, 2020-02-24:
I have the same issue on an iPhone 11 Pro iOS 13.2.3

toams69, 2020-02-28:
Same issue here.
Was working fine before the update of IOS.

toams69, 2020-02-28:
Fixed on my side by adding the key NSBluetoothAlwaysUsageDescription in the info.plist

https://developer.apple.com/documentation/bundleresources/information_property_list/nsbluetoothalwaysusagedescription

manojdcoder, 2020-03-01:
`NSBluetoothAlwaysUsageDescription` was recently introduced in iOS 13. Without that your app could crash.

I just ran a test on iOS 13.2.3, it worked just fine.

luca-mayven, 2020-03-01:
Thanks a lot @toams69 and @manojdcoder !

It works fine now that I added the key as well.

Now it seems to be scanning correctly but the list looks empty. 

In any case the original issue is solved.

Thanks

Luca

Joebh, 2020-03-20:
Using 3.0.0-beta.10 I'm not able to see any bluetooth devices while scanning on ios 12 iphone 5s, anyone able to get it working with that version?

manojdcoder, 2020-03-20:
@Joebh Were you able to test it on any other devices, perhaps one with iOS 13?

farfromrefug, 2020-03-20:
I am using the latest beta in production on iOS and Android. It is working fine everywhere. Check you logs you might be missing permissions

Joebh, 2020-03-23:
@manojdcoder verified it's working ios13 ty, haven't found the issue yet for 12.

manojdcoder, 2020-03-23:
I have to setup iOS 12, I will update when I do so.

closed 2020-08-27 by farfromrefug