nativescript-community/ble #170: v2/v3 Beta Changelog Question

brandonfraune, 2020-01-15:
@EddyVerbruggen @farfromrefug 

**TLDR**; Can you help me understand the changes between v1.3.1, v2.0.0-beta(.27), and v3.0.0-beta(.9)? I couldn't find this information on the NPM site, and this repo doesn't appear to have any tags (or changelog info) for beta releases. 

The {N}6 project I'm working on is dependent on this library, and we're having some issues establishing Bluetooth connections, that we didn't experience when we were on {N}5 with v1.3.1 of this package. It's difficult to reproduce, so to help me understand if these issues are related to our update to the v2-beta, I'd like to get a better understanding of what's happening in the v2-beta and v3-beta.

1. What were the goals of v2? (I presume support for {N}6 and AndroidX?)
2. What's keeping v2 in beta? (is the [v2 milestone](https://github.com/EddyVerbruggen/nativescript-bluetooth/milestone/4) still active?)
3. What updates prompted a major release change to v3?
4. What's keeping v3 in beta? (will a v3 milestone be created?)

Apologies if this isn't the proper place to ask these questions, but I appreciate your support.

farfromrefug, 2020-01-15:
@brandonfraune the fact that i went for v3 is that v3 requires @nativescript and not tns-core-modules which is a major change.
Now i cant make the choice to make the v3 the "master" one. @EddyVerbruggen have to make that choice.
But i am thinking about creating a new plugin altogether with v3. Someting like ```nativescript-ble```. It would also help with people constantly asking to Bluetooth 2.0 support ;)

brandonfraune, 2020-01-20:
Thanks for your response. Given that the beta nativescript-bluetooth packages available on the NPM site are not available within this repo, is there a plan to add a CHANGELOG.md to the NPM packages going forward? This would help with locating change details, without needing to chase down commits on multiple repos.

msn444, 2020-01-21:
Count me too as very confused by the state of this plugin. I see there is a v3 mentioned in various issues, and available in beta on NPM, but there is no evidence of its existence in the repo. Where/what is it?

Then there's v2, which has long had a branch in the repo but no documentation.

Is this plugin still a going concern?

farfromrefug, 2020-01-21:
For you guys here in the repo look for the v2 branch! But yes no changelog or doc. Though the typings are up to date which should be enough. Any PRs is welcome. BTW I am about (I think) to give up on this plug-in and write a new one (adapter for a shared code with nodejs)

farfromrefug, 2020-01-21:
@msn444 btw I still heavily maintain the v3 as it is already used in production and its getting big

msn444, 2020-01-21:
Ok thanks @farfromrefug . But where can I find v3 in the repo? I don't see any tags or branch relating to it.

prabudevarrajan, 2020-01-23:
Hi @farfromrefug @msn444, I am trying to use bluetooth in my project and I find it hard to make it work on both android and ios. First, I tried to use the github repo 

https://github.com/EddyVerbruggen/nativescript-bluetooth

But it worked on Android and not on iOS. 

Then, I looked at the comment and tried to use the following branch https://github.com/msn444/nativescript-bluetooth, but I am getting the below errors.

Can you all point me to right github branch which has the latest version of the bluetooth which works both on iOS and Android

$ git clone git@github.com:msn444/nativescript-bluetooth.git
$ cd nativescript-bluetooth/demo-ng
$ nativescript --version
6.3.0
$ tns migrate
$ tns debug android

```

$ tns debug android

Improve your app by switching to "markingMode:none".

With the upcoming NativeScript 7.0 the "markingMode:none" will become the only marking mode supported by the Android Runtime.

? Do you want to switch your app to the recommended "markingMode:none"?
More info about the reasons for this change can be found in the link below:
https://www.nativescript.org/blog/markingmode-none-is-official-boost-android-performance-while-avoiding-memory-issues Yes
Searching for devices...
Copying template files...
Platform android successfully added. v6.0.0
Preparing project...
File change detected. Starting incremental webpack compilation...

webpack is watching the files…

Hash: 6a9b93dfb8de2f51c869
Version: webpack 4.27.1
Time: 15774ms
Built at: 01/22/2020 5:13:29 PM
                  Asset      Size   Chunks             Chunk Names
              bundle.js  8.71 KiB   bundle  [emitted]  bundle
           package.json  78 bytes           [emitted]  
res/signal-strength.png  2.82 KiB           [emitted]  
             runtime.js  71.4 KiB  runtime  [emitted]  runtime
    tns-java-classes.js   0 bytes           [emitted]  
              vendor.js  3.58 MiB   vendor  [emitted]  vendor
Entrypoint bundle = runtime.js vendor.js bundle.js
[./app.css] 1.42 KiB {bundle} [built]
[./main.ts] 1.17 KiB {bundle} [built]
[./package.json] 80 bytes {bundle} [optional] [built]
    + 172 hidden modules

ERROR in app/components/characteristicList.component/characteristicList.component.ts(34,40): error TS2339: Property 'services' does not exist on type 'IPeripheral'.
app/components/characteristicList.component/characteristicList.component.ts(35,87): error TS2339: Property 'UUID' does not exist on type 'IService'.
app/components/characteristicList.component/characteristicList.component.ts(35,119): error TS2339: Property 'name' does not exist on type 'IService'.
app/components/peripheralList.component/peripheralList.component.ts(21,65): error TS2339: Property 'UUID' does not exist on type 'IPeripheral'.
app/components/svcList.component/svcList.component.ts(64,70): error TS2339: Property 'UUID' does not exist on type 'IPeripheral'.
app/services/bluetooth.service.ts(5,76): error TS2307: Cannot find module 'nativescript-bluetooth'.
app/services/bluetooth.service.ts(94,18): error TS2322: Type '{ UUID: string; name: string; RSSI: number; services: undefined[]; }' is not assignable to type 'IPeripheral'.
  Object literal may only specify known properties, and 'UUID' does not exist in type 'IPeripheral'.
app/services/bluetooth.service.ts(99,13): error TS2322: Type '{ UUID: string; name: string; RSSI: number; services: undefined[]; }' is not assignable to type 'IPeripheral'.
  Object literal may only specify known properties, and 'UUID' does not exist in type 'IPeripheral'.
app/services/bluetooth.service.ts(112,24): error TS2339: Property 'services' does not exist on type 'IPeripheral'.
app/services/bluetooth.service.ts(113,24): error TS2339: Property 'services' does not exist on type 'IPeripheral'.
app/services/bluetooth.service.ts(115,25): error TS2339: Property 'characteristics' does not exist on type 'IService'.
app/services/bluetooth.service.ts(116,25): error TS2339: Property 'characteristics' does not exist on type 'IService'.
app/services/bluetooth.service.ts(129,65): error TS2339: Property 'UUID' does not exist on type 'IPeripheral'.
app/services/bluetooth.service.ts(144,67): error TS2339: Property 'UUID' does not exist on type 'IPeripheral'.
app/services/bluetooth.service.ts(154,40): error TS2339: Property 'UUID' does not exist on type 'IPeripheral'.
[... 6 lines cut ...]
app/services/bluetooth.service.ts(306,77): error TS2339: Property 'UUID' does not exist on type 'IPeripheral'.
app/services/bluetooth.service.ts(307,48): error TS2339: Property 'UUID' does not exist on type 'IPeripheral'.
app/services/bluetooth.service.ts(330,73): error TS2339: Property 'UUID' does not exist on type 'IPeripheral'.
app/services/bluetooth.service.ts(331,56): error TS2339: Property 'UUID' does not exist on type 'IService'.
app/services/bluetooth.service.ts(332,52): error TS2339: Property 'UUID' does not exist on type 'ICharacteristic'.
app/services/bluetooth.service.ts(343,66): error TS2339: Property 'UUID' does not exist on type 'ICharacteristic'.
app/services/bluetooth.service.ts(380,73): error TS2339: Property 'UUID' does not exist on type 'IPeripheral'.
app/services/bluetooth.service.ts(381,56): error TS2339: Property 'UUID' does not exist on type 'IService'.
app/services/bluetooth.service.ts(382,52): error TS2339: Property 'UUID' does not exist on type 'ICharacteristic'.
app/services/bluetooth.service.ts(390,67): error TS2339: Property 'UUID' does not exist on type 'ICharacteristic'.
[... cut ...]
app/services/bluetooth.service.ts(466,76): error TS2339: Property 'UUID' does not exist on type 'ICharacteristic'.
app/services/bluetooth.service.ts(480,73): error TS2339: Property 'UUID' does not exist on type 'IPeripheral'.
app/services/bluetooth.service.ts(481,56): error TS2339: Property 'UUID' does not exist on type 'IService'.
app/services/bluetooth.service.ts(482,52): error TS2339: Property 'UUID' does not exist on type 'ICharacteristic'.
app/services/bluetooth.service.ts(490,75): error TS2339: Property 'UUID' does not exist on type 'ICharacteristic'.

```

msn444, 2020-01-23:
@prabudevarrajan I wouldn't use the msn444 branch; I don't even remember why I created that.

Clone the v2 branch: `git clone https://github.com/EddyVerbruggen/nativescript-bluetooth.git --branch v2` 
I haven't tried the angular demo but the standard one seems to work.
I currently use the 2.0.0-beta27 release (the most recent 2.0.0 release), so run: 
`tns plugin add nativescript-bluetooth@2.0.0-beta27` in the demo folder.
As I noted in #171 , on Android you may need to use nativescript-permissions to request ACCESS_FINE_LOCATION permissions, ie:
`const permissions = require("nativescript-permissions");
permissions.requestPermission(android.Manifest.permission.ACCESS_FINE_LOCATION, "I need permissions").then(() => { 
\\ startScanning, etc.
})`

brandonfraune, 2020-01-23:
@farfromrefug Thanks for your work, and for helping answer my questions about the state of this plugin. I don't have any other questions beyond this, so I'm going to close this thread before it gets too off-topic.

closed 2020-01-23 by brandonfraune