nativescript-community/ble #173: Scan BLE on Android 10

detonimarco, 2020-01-27:
Hi,
I'm using version 3.0.0-beta.9 with nativescript 6.2.0 and nativescript-angular 8.20.3
Doing scan on Android 10 the procedure doesn't find devices.
On Android 9 it works.

Is it a configuration/permission I need to set for Android 10?

Regards

farfromrefug, 2020-01-27:
@detonimarco https://stackoverflow.com/questions/58428763/android-10-not-working-with-ble-bluetooth-scanning

detonimarco, 2020-01-27:
Thanks.
I googled but I missed it....

niko01360, 2020-02-12:
Hello,

I have the same problem. On my android 10, I have no results in my scan.
However, I did add the: ACCESS_BACKGROUND_LOCATION in the manifest file.
But impossible to display the available bluetooth.
I have no problem on Android < 10

EikiProg, 2020-02-17:
Hello, I have the same problem on android 10 but work fine on other version 9 and lower

I have this in my maniufest:
```
<uses-permission android:name="android.permission.BLUETOOTH" />
<uses-permission android:name="android.permission.BLUETOOTH_ADMIN" />
<uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />
<uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" />
<uses-permission android:name="android.permission.ACCESS_BACKGROUND_LOCATION" />
```

manojdcoder, 2020-02-19:
While [targeting Android 10 ](https://developer.android.com/about/versions/10/privacy/changes#location-telephony-bluetooth-wifi) your app must have `ACCESS_FINE_LOCATION` permission. 

The plugin requests for `ACCESS_COARSE_LOCATION` which seems to be the reason for this issue. A workaround is to request for the permission that permission at runtime before calling `startScanning(...)`.

detonimarco, 2020-02-19:
thanks @manojdcoder adding ACCESS_FINE_LOCATION the scan works
I also added 
    permissions.requestPermissions([android.Manifest.permission.ACCESS_FINE_LOCATION,
        android.Manifest.permission.ACCESS_FINE_LOCATION],
            'App Needs The Following permissions to access BLE device');

EikiProg, 2020-02-20:
Ok it's working now, I don't know why but we need to have the GPS enabled for it to work.

It would be good to return an error for SDK 29+ when GPS isn't enable in the error promise of startScanning to know why it's not working, because we can't enable it with the method in this plugin. Like this we can let the user know that they to enable it manualy.

closed 2020-08-27 by farfromrefug

st920228, 2020-11-12:
//gradle
dependencies {
implementation 'no.nordicsemi.android.support.v18:scanner:1.4.3'
}
//import
import no.nordicsemi.android.support.v18.scanner.BluetoothLeScannerCompat;
import no.nordicsemi.android.support.v18.scanner.ScanCallback;
import no.nordicsemi.android.support.v18.scanner.ScanResult;
import no.nordicsemi.android.support.v18.scanner.ScanSettings;
//init scan setting
final ScanSettings settings = new ScanSettings.Builder().setReportDelay(1000).setScanMode(ScanSettings.SCAN_MODE_LOW_LATENCY).build();
//start scan
scanner.startScan(null, settings, scanCallback);
//scan callback
private final ScanCallback scanCallback = new ScanCallback() {
        @Override
        public void onScanResult(final int callbackType, @NonNull final ScanResult result) {
            super.onScanResult(callbackType, result);
        }
        public void onBatchScanResults(@NonNull final List<ScanResult> results) {
            for (final ScanResult result : results) {
                //result.getDevice() is scanned device
            }
        }
        @Override
        public void onScanFailed(final int errorCode) {
            // should never be called
        }
    };

ray007, 2021-01-18:
Seems the README needs an update.