nativescript-community/ble #174: IOS - Unable to connect using version > 1.3.1

toams69, 2020-02-23:
Hi,
I've  upgrade the version of nativescript-bluetooth from 1.3.1 to 3.0.0-beta.10.
I
All was ok with the version 1.3.1, but since this upgrade I am not able to connect an IOS system to the ble device. 
The scan method is working well but an error occurs during the connection part.

```
'[nativescript-bluetooth]' 'connect' '----' '82855044-1508-0b20-2f09-39da76518c45'
vendor.js:68366 '[nativescript-bluetooth]' 'connect' '---- peripheral found' {}
vendor.js:68366 '[nativescript-bluetooth]' 'connect' '---- Connecting to peripheral with UUID:' '82855044-1508-0b20-2f09-39da76518c45' { [super]: {}, _owner: {}, subDelegates: [ [length]: 0 ] } {}
vendor.js:68366 '[nativescript-bluetooth]' 'connect' '----about to connect:' '82855044-1508-0b20-2f09-39da76518c45' { [super]: {},
  _owner: {},
  subDelegates: 
   [ { centralManagerDidConnectPeripheral: [Object],
       centralManagerDidFailToConnectPeripheralError: [Object] },
     [length]: 1 ] } {}
vendor.js:68366 '[nativescript-bluetooth]' '----- CBCentralManagerDelegateImpl centralManager:didConnectPeripheral: <CBPeripheral: 0x2832a5540, identifier = 82855044-1508-0B20-2F09-39DA76518C45, name = testdevice, state = connected>'
vendor.js:68366 '[nativescript-bluetooth]' "'CBPeripheralDelegateImpl.initWithOwner: [object Object]'"
vendor.js:68366 '[nativescript-bluetooth]' '----- CBCentralManagerDelegateImpl centralManager:didConnectPeripheral, let\'s discover service'
vendor.js:68366 '[nativescript-bluetooth]' 'discoverServices' '---- peripheralUUID:82855044-1508-0b20-2f09-39da76518c45'
vendor.js:68366 '[nativescript-bluetooth]' 'CBPeripheralDelegateImpl.peripheralDidDiscoverServices ---- peripheral: <CBPeripheral: 0x2832a5540, identifier = 82855044-1508-0B20-2F09-39DA76518C45, name = testdevice, state = connected>, null, <CBPeripheralDelegateImpl: 0x2801a7c50>'
vendor.js:68366 '[nativescript-bluetooth]' 'discoverCharacteristics' '---- peripheralUUID:82855044-1508-0b20-2f09-39da76518c45 serviceUUID:ffb0'
vendor.js:68366 '[nativescript-bluetooth]' 'CBPeripheralDelegateImpl.peripheralDidDiscoverCharacteristicsForServiceError ---- peripheral: <CBPeripheral: 0x2832a5540, identifier = 82855044-1508-0B20-2F09-39DA76518C45, name = testdevice, state = connected>, service: <CBService: 0x2816b0440, isPrimary = YES, UUID = 0000FFB0-0000-1000-8000-00805F9B34FB>, error: null'
```

I need to upgrade native script to make it working on android.

Hope you can help

Regards
Thomas

manojdcoder, 2020-02-24:
@toams69 What's the error, is it a crash Or exception? Do you see any error description? Can you share a sample project?

toams69, 2020-02-24:
@manojdcoder there is no crash, the connect promise is not resolved nor rejected. I had to put the plugin in debug mode to see the log provided.

I will share the full logs tonight if necessary. 
The code is very basic after scanning I am using the UUID to connect to the peripheral then this error occurs.

toams69, 2020-02-24:
The same code is working fine on Android btw

methompson, 2020-03-11:
I too am encountering errors concerning this plugin on iOS. Here's the gist of my code:

```
import { Bluetooth } from "nativescript-bluetooth";
const bluetooth = new Bluetooth();

bluetooth.isBluetoothEnabled()
        .then((enabled) => {
          console.log('Enabled? ' + enabled);
        })
        .catch((err) => {
          console.log(err);
        });
```

I get two separate errors. I get an error in the Xcode Console from the app:

*** Assertion failure in -[CBCentralManager initWithDelegate:queue:options:], /BuildRoot/Library/Caches/com.apple.xbs/Sources/MobileBluetoothFramework/MobileBluetooth-133.3/CoreBluetooth/CoreBluetooth/CBCentralManager.m:230

I also get an NS error in the JS Console:

Error: State restoration of CBCentralManager is only allowed for applications that have specified the "bluetooth-central" background mode

I'm not quite sure how I can get around this in iOS without just using a forked version of v1.3.1

manojdcoder, 2020-03-11:
@methompson Did you try v3.0.0-beta.10 which is latest on NPM?

methompson, 2020-03-11:
@manojdcoder Yes. That's what I have installed.

manojdcoder, 2020-03-11:
I tested it a few days ago and it was working on my end. Would you like to share a sample project if the issue still persists on your end.

methompson, 2020-03-11:
@manojdcoder The sample I wrote above is literally what I'm using to test the code and the errors I wrote are what I get on a real phone.

manojdcoder, 2020-03-11:
There could be other factors, easy to debug when you share the project like what runtime you are using, changes on info.plist (if you had not checked https://github.com/EddyVerbruggen/nativescript-bluetooth/issues/169#issuecomment-593029638) etc., 

All that I could say without a sample project, it works on my end.

farfromrefug, 2020-03-11:
Why dont you simply add bluetooth-central background mode if it asks you to?

methompson, 2020-03-11:
@farfromrefug because I didn’t have to enable background-central mode for v1.

farfromrefug, 2020-03-12:
But did you try ? The log is telling you to do so

methompson, 2020-03-12:
@farfromrefug yes, I did, eventually. Switching that on made my project work.

My problem was that without knowing anything about background mode, the logs didn’t make much sense.

manojdcoder, 2020-03-12:
Apple changes their requirements about permissions and stuffs in every version. But I'm still not sure why you needed background mode, except if you had enabled Bluetooth usage while app is in background.

closed 2020-08-27 by farfromrefug