nativescript-community/ble #179: Beta Documentation

methompson, 2020-04-11:
I've been using the beta version of this plugin for a bit and I'm noticing a fair amount of divergence between the Readme in the readme-2x branch and the way the plugin actually works.

Is there a more up-to-date document? If not, @EddyVerbruggen do you have a problem with a PR with some updated the documentation based upon some of the things I've noticed?

Here are a few points I've noticed:

The documentation states: 
```
import { Bluetooth } from 'nativescript-bluetooth';

const bluetooth = new Bluetooth();
```

But, the documentation doesn't state how an app handles multiple instances of a bluetooth object. For instance, if we have a parent that makes a bluetooth object, then a child component makes a bluetooth object, that second bluetooth object will effectively "steal" the ability to access bluetooth from the parent object, preventing the object from reading or writing. I've had to create a single bluetooth object for an entire app and put it into a module so that all components can read/write without stealing from other components. I discovered this because I have a menu UI element that exists on top of the rest of the app that allows a user to connect a Bluetooth device. Other pages that exist within a frame (a child component of the menu element) use bluetooth and they would frequently steal use of bluetooth from other existing objects that persist.

The documentation states:

> ### enable (Android only)
> The promise will be rejected on iOS

The Promise actually resolves on iOS.

The documentation states in the write method section:

```
bluetooth
  .write({
    peripheralUUID: '34134-5453-4453-54545',
    serviceUUID: '180e',
    characteristicUUID: '3424-45234-34324-2343',
    value: '0x01' // a hex 1
  })
```

Hex values written as a string used to work in v1, but in v3 beta, an actual hex value (e.g. `0x01` without quotation marks) is required.

The above were discovered in the normal course of upgrading from v1 to v3.

farfromrefug, 2020-04-11:
@methompson i am the one maintaining the v2 branch. PRs are more than welcome!

methompson, 2020-04-11:
@farfromrefug Just so that I'm clear, is the readme-2x the most up-to-date documentation?

farfromrefug, 2020-04-11:
Yes

farfromrefug, 2020-08-27:
readme fixed for the new `@nativescript-community/ble`

closed 2020-08-27 by farfromrefug