nativescript-community/ble #180: Filters?

Stan92, 2020-04-16:
Hi,

Nativescript-bluetooth version : 3.0.0-beta.10

Using a filter to the startScanning method, I can't find my devices.
Do I miss something?

```
       const filters=[{serviceUUID:'5001'}];
        const seconds=3;
        bluetooth.startScanning(
            {
             filters,
             seconds,
              onDiscovered:(peripheral:Peripheral)=>{
                   console.log(peripheral);
              },
              skipPermissionCheck:false
        }).then(
            (res)=>{
                console.log("ble scan terminé");
            },
            (err)=>{
                console.log("err",err,true);
            });

```

manojdcoder, 2020-04-17:
What kind of device you are trying to connect? Where did you find the service UUID from? `5001` is something not listed in the [spec](https://www.bluetooth.com/specifications/gatt/services/).

Stan92, 2020-04-17:
It’s a esp32 part with a customized code

manojdcoder, 2020-04-17:
Do you find the specific device on list if you do not apply filters?

Stan92, 2020-04-17:
Yes I can see it, connect, read the services and characteristics 
Everything works except the filters
Btw thanks for the package

manojdcoder, 2020-04-17:
Did you test on both iOS & Android? Is it platform specific?

Stan92, 2020-04-17:
I only tested on Android.

manojdcoder, 2020-04-17:
Are you able to perform a read / write with the service id?

Stan92, 2020-04-17:
Yes for the read (I don't have a write)
Everything perfectly works except the filters :) 
I have also noticed, the Peripheral doesn't have the state property (within the interface).
But if I log the object, I get these following properties
 type,UUID,name,RSSI,localName,state,manufacterId,advertismentData

Stan92, 2020-05-01:
Hi,
It seems it doesn't work too with iOS (probably I missed something).
I also tried with a "standard" service UUID (coming from my MacBook Pro).