nativescript-community/ble #186: Remove onDisconnect event

senner007, 2020-08-07:
Hi :)

It could be nice if the onDisconnect event could be removed.
If I redirect to an error page in the event I wouldn't want that to happen if I disconnect after a successful read.

So something like this:
```js
  public removeDisconnectEvent(UUID): void {

    if (!this.connections[UUID]) {
      throw new Error("Peripheral wasn't connected")
    }
    this.connections[UUID].onDisconnected = null;
  }
```

farfromrefug, 2020-08-07:
@senner007 does not need to be handle in this module. Your logic can and should be handled in your app.
I do that myself without any issue.

senner007, 2020-08-07:
Yes of course it can be done. Anything is possible by adding ifs and elses

farfromrefug, 2020-08-07:
Yes but not everything cas its place in the plugin. Some intelligence should remain on the app side and I think this is one of those.

closed 2020-08-27 by farfromrefug