nativescript-community/ble #187: Build error "An accessor cannot be declared in an ambient context."

martintschemernjak, 2020-08-07:
Hi,

I have NativeScript 6.5.1 and use the nativescript-bluetooth Version 3.0.0-beta.11 and i also tried 3.0.0-beta.12. 
On both i get the following compiletime error:

```
ERROR in node_modules/nativescript-bluetooth/bluetooth.common.d.ts:24:9 - error TS1086: An accessor cannot be declared in an ambient context.

24     set debug(value: boolean);
```

If I go to this file and comment the line out, all is good and it compiles. It also works as it is supposed to.
Problem is with any CI Tool... I can not build the app on jenkins or circleCI because of that. I can not manually edit the file there. 

Here is also my package.json:
```
{
  "nativescript": {
    "id": "org.nativescript.testapp",
    "tns-android": {
      "version": "6.5.1"
    }
  },
  "description": "APP DESC",
  "license": "SEE LICENSE IN <your-license-filename>",
  "repository": "<fill-your-repository-here>",
  "scripts": {
    "lint": "tslint \"src/**/*.ts\""
  },
  "dependencies": {
    "@angular/animations": "^8.2.14",
    "@angular/common": "^8.2.14",
    "@angular/compiler": "^8.2.14",
    "@angular/core": "^8.2.14",
    "@angular/forms": "^8.2.14",
    "@angular/platform-browser": "^8.2.14",
    "@angular/platform-browser-dynamic": "^8.2.14",
    "@angular/router": "^8.2.14",
    "@nativescript/theme": "^2.3.2",
    "@nstudio/nativescript-floatingactionbutton": "^2.0.0",
    "@types/text-encoding": "0.0.35",
    "moment": "^2.24.0",
    "nativescript-angular": "^8.21.0",
    "nativescript-bluetooth": "3.0.0-beta.12",
    "nativescript-cscreenshot": "1.0.2",
    "nativescript-feedback": "^1.3.12",
    "nativescript-geolocation": "^5.1.0",
    "nativescript-localize": "^4.2.1",
    "nativescript-modal-datetimepicker": "^1.2.2",
    "nativescript-permissions": "^1.3.8",
    "nativescript-plugin-firebase": "^10.5.0",
    "nativescript-screen-orientation": "2.0.0",
    "nativescript-share-file-enduco": "1.0.7",
    "nativescript-ui-calendar": "^6.0.0",
    "nativescript-ui-chart": "^6.0.0",
    "nativescript-ui-dataform": "^6.0.0",
    "nativescript-ui-gauge": "^6.0.0",
    "nativescript-ui-listview": "^8.0.1",
    "nativescript-ui-sidedrawer": "8.0.0",
    "reflect-metadata": "~0.1.13",
    "rxjs": "^6.5.4",
    "text-encoding": "0.7.0",
    "tns-core-modules": "~6.4.1",
    "underscore": "^1.9.2",
    "zone.js": "^0.9.1"
  },
  "devDependencies": {
    "@angular/compiler-cli": "^8.2.14",
    "@nativescript/schematics": "^1.0.0",
    "@ngtools/webpack": "^8.3.25",
    "codelyzer": "^5.2.1",
    "nativescript-dev-webpack": "~1.5.0",
    "node-sass": "^4.13.1",
    "tslint": "~5.17.0",
    "typescript": "^3.5.3"
  },
  "readme": "NativeScript Application",
  "author": "AUTHOR"
}
```
Does anybody know what might be the problem here? Any help would be very much appreciated.
Thank you very much,
best regards,
Martin

farfromrefug, 2020-08-08:
@martintschemernjak that has been fixed some time ago. Are you sure you are up to date?

martintschemernjak, 2020-08-08:
@farfromrefug Yes, I think so.
I also deleted the node_modules folder several times and as you can see i have   "nativescript-bluetooth": "3.0.0-beta.12" referenced in my package.json.

farfromrefug, 2020-08-08:
OK maybe I need to publish another beta

martintschemernjak, 2020-08-12:
yes, that would be great.
When do you think you could do that?

farfromrefug, 2020-08-27:
it should be published

closed 2020-08-27 by farfromrefug