nativescript-community/ble #188: Class constructor Observable cannot be invoked without 'new'

psorin, 2020-08-18:
Hi,

I'm using nativescript 6.5.3 and nativescript-bluetooth 3.0.0-beta.10.

when running the application I receive the error:

System.err: TypeError: Class constructor Observable cannot be invoked without 'new'
System.err: File: (file: node_modules\nativescript-observable\observable.js:6:0)
System.err: StackTrace:
System.err: Observable(file: node_modules\nativescript-observable\observable.js:6:0)
System.err:     at BluetoothCommon(file: node_modules\nativescript-bluetooth\bluetooth.common.js:106:0)
System.err:     at Bluetooth(file: node_modules\nativescript-bluetooth\bluetooth.android.js:1001:0)
System.err:     at ./app/shared/services/bluetooth.service.ts(file: src\app\shared\services\bluetooth.service.ts:4:18)

This is the package.json:

{
  "nativescript": {
    "id": "org.nativescript.sampleapp",
    "tns-ios": {
      "version": "6.5.2"
    },
    "tns-android": {
      "version": "6.5.3"
    }
  },
  "description": "NativeScript Application",
  "license": "SEE LICENSE IN <your-license-filename>",
  "repository": "<fill-your-repository-here>",
  "scripts": {
    "clean": "npx rimraf hooks node_modules package-lock.json platforms",
    "ngcc": "ngcc --properties es2015 module main --first-only",
    "postinstall": "npm run ngcc"
  },
  "dependencies": {
    "@angular/animations": "~10.0.0",
    "@angular/common": "~10.0.0",
    "@angular/compiler": "~10.0.0",
    "@angular/core": "~10.0.0",
    "@angular/forms": "~10.0.0",
    "@angular/platform-browser": "~10.0.0",
    "@angular/platform-browser-dynamic": "~10.0.0",
    "@angular/router": "~10.0.0",
    "@nativescript/angular": "rc",
    "@nativescript/core": "rc",
    "@nativescript/theme": "~2.3.0",
    "nativescript-bluetooth": "^3.0.0-beta.10",
    "nativescript-geolocation": "^5.1.0",
    "nativescript-secure-storage": "^2.6.1",
    "reflect-metadata": "~0.1.12",
    "rxjs": "^6.5.0",
    "zone.js": "~0.10.3"
  },
  "devDependencies": {
    "@angular/compiler-cli": "~10.0.0",
    "@nativescript/types": "rc",
    "@nativescript/webpack": "~2.1.0",
    "@ngtools/webpack": "~10.0.0",
    "typescript": "~3.9.0"
  },
  "readme": "NativeScript Application"
}
Does anybody know what might be the problem here? Any help would be very much appreciated. 

Thank you very much
Sorin

JakeAi, 2020-08-24:
@psorin same problem. It just started after I updated to angular 10

farfromrefug, 2020-08-24:
@psorin @JakeAi You are not using nativescript 6.5.3 you are using the RC version for N 7.0
`nativescript-bluetooth` is not compatible yet. When it is i will release it under `@nativescript-community/ble`

farfromrefug, 2020-08-27:
@psorin @JakeAi `@nativescript-community/ble` has been published for N 7.0!

closed 2020-08-27 by farfromrefug