nativescript-community/ble #189: connect problem iOS

AvigarD611, 2020-08-27:
Hello, today I have installed this plugin and I have a problem, when connect to bluetooth device. Scan is working. Here is the error:

***** Fatal JavaScript exception - application has been terminated. *****
Native stack trace:
1   0x102f36928 NativeScript::reportFatalErrorBeforeShutdown(JSC::ExecState*, JSC::Exception*, bool)
2   0x102f7102c NativeScript::FFICallback<NativeScript::ObjCMethodCallback>::ffiClosureCallback(ffi_cif*, void*, void**, void*)
3   0x103ade7f8 ffi_closure_SYSV_inner
4   0x103ae01b4 .Ldo_closure
5   0x1bd9438dc <redacted>
6   0x1bd94025c <redacted>
7   0x1bd93aa74 <redacted>
8   0x1bd95557c <redacted>
9   0x1bd94c18c <redacted>
10  0x1a91b2134 <redacted>
11  0x1a91b35ac <redacted>
12  0x1a91b9a64 <redacted>
13  0x1a91ba4cc <redacted>
14  0x1a91bf6c4 _dispatch_main_queue_callback_4CF
15  0x1a948c8d4 <redacted>
16  0x1a948758c <redacted>
17  0x1a9486bc8 CFRunLoopRunSpecific
18  0x1b38685cc GSEventRunModal
19  0x1ad639744 UIApplicationMain
20  0x103ae0044 ffi_call_SYSV
21  0x103addf1c ffi_call_int
22  0x103adda18 ffi_call
23  0x102ef946c NativeScript::FunctionWrapper::call(JSC::ExecState*)
24  0x103aaacf0 llint_entry
25  0x103aa8108 llint_entry
26  0x103aa8108 llint_entry
27  0x103aa8108 llint_entry
28  0x103aa8108 llint_entry
29  0x103aa8108 llint_entry
30  0x103aa8108 llint_entry
31  0x103aa8060 llint_entry
JavaScript stack trace:
peripheralDidDiscoverServices(file: app/src/bluetooth.ios.ts:1375:64)
at file:///app/vendor.js:364:40
at forEach([native code])
at file:///app/vendor.js:362:30
at UIApplicationMain([native code])
at run(file: node_modules/@nativescript/core/application/application.ios.js:312:26)
at file: node_modules/nativescript-vue/dist/index.js:14050:2
at file:///app/bundle.js:546:10
at ./app.js(file:///app/bundle.js:550:34)
at __webpack_require__(file: app/webpack/bootstrap:750:0)
at checkDeferredModules(file: app/webpack/bootstrap:43:0)
at webpackJsonpCallback(file: app/webpack/bootstrap:30:0)
at anonymous(file:///app/bundle.js:2:61)
at evaluate([native code])
at moduleEvaluation([native code])
at [native code]
at asyncFunctionResume([native code])
at [native code]
at promiseReactionJob([native code])
JavaScript error:
file: app/src/bluetooth.ios.ts:1375:64: JS ERROR TypeError: undefined is not an object (evaluating '_nativescript_core_utils_native_helper__WEBPACK_IMPORTED_MODULE_1__["iOSNativeHelper"].collections')
*** JavaScript call stack:
(
0   UIApplicationMain@[native code]
1   run@file: node_modules/@nativescript/core/application/application.ios.js:312:26
2   @file: node_modules/nativescript-vue/dist/index.js:14050:2
3   @file:///app/bundle.js:546:10
4   ./app.js@file:///app/bundle.js:550:34
5   __webpack_require__@file: app/webpack/bootstrap:750:0
6   checkDeferredModules@file: app/webpack/bootstrap:43:0
7   webpackJsonpCallback@file: app/webpack/bootstrap:30:0
8   anonymous@file:///app/bundle.js:2:61
9   evaluate@[native code]
10  moduleEvaluation@:1:11
11  @:2:1
12  asyncFunctionResume@:1:11
13  @:24:9
14  promiseReactionJob@:1:11
)
*** Terminating app due to uncaught exception 'NativeScript encountered a fatal error: TypeError: undefined is not an object (evaluating '_nativescript_core_utils_native_helper__WEBPACK_IMPORTED_MODULE_1__["iOSNativeHelper"].collections')
at
peripheralDidDiscoverServices(file: app/src/bluetooth.ios.ts:1375:64)
at file:///app/vendor.js:364:40
at forEach([native code])
at file:///app/vendor.js:362:30
at UIApplicationMain([native code])
at run(file: node_modules/@nativescript/core/application/application.ios.js:312:26)
at file: node_modules/nativescript-vue/dist/index.js:14050:2
at file:///app/bundle.js:546:10
at ./app.js(file:///app/bundle.js:550:34)
at __webpack_require__(file: app/webpack/bootstrap:750:0)
at checkDeferredModules(file: app/webpack/bootstrap:43:0)
at webpackJsonpCallback(file: app/webpack/bootstrap:30:0)
at anonymous(file:///app/bundle.js:2:61)
at evaluate([native code])
at moduleEvaluation([native code])
at [native code]
at asyncFunctionResume([native code])
at [native code]
at promiseReactionJob([native code])
', reason: '(null)'
*** First throw call stack:
(0x1a9511344 0x<\M-b\M^@\M-&>
NativeScript caught signal 6.
Native Stack:
1   0x102f7fc7c sig_handler(int)
2   0x1a9215404 <redacted>
3   0x1a9218948 pthread_kill
4   0x1a91a7ba4 abort
5   0x1a92cbf60 <redacted>
6   0x1a92bd92c <redacted>
7   0x1a9226fa8 <redacted>
8   0x1a92cb40c <redacted>
9   0x1a92cdecc __cxa_get_exception_ptr
10  0x1a92cde84 <redacted>
11  0x1a9226df0 <redacted>
12  0x102f36e00 NativeScript::reportFatalErrorBeforeShutdown(JSC::ExecState*, JSC::Exception*, bool)
13  0x102f7102c NativeScript::FFICallback<NativeScript::ObjCMethodCallback>::ffiClosureCallback(ffi_cif*, void*, void**, void*)
14  0x103ade7f8 ffi_closure_SYSV_inner
15  0x103ae01b4 .Ldo_closure
16  0x1bd9438dc <redacted>
17  0x1bd94025c <redacted>
18  0x1bd93aa74 <redacted>
19  0x1bd95557c <redacted>
20  0x1bd94c18c <redacted>
21  0x1a91b2134 <redacted>
22  0x1a91b35ac <redacted>
23  0x1a91b9a64 <redacted>
24  0x1a91ba4cc <redacted>
25  0x1a91bf6c4 _dispatch_main_queue_callback_4CF
26  0x1a948c8d4 <redacted>
27  0x1a948758c <redacted>
28  0x1a9486bc8 CFRunLoopRunSpecific
29  0x1b38685cc GSEventRunModal
30  0x1ad639744 UIApplicationMain
31  0x103ae0044 ffi_call_SYSV
JS Stack:
UIApplicationMain([native code])
at run(file: node_modules/@nativescript/core/application/application.ios.js:312:26)
at file: node_modules/nativescript-vue/dist/index.js:14050:2
at file:///app/bundle.js:546:10
at ./app.js(file:///app/bundle.js:550:34)
at __webpack_require__(file: app/webpack/bootstrap:750:0)
at checkDeferredModules(file: app/webpack/bootstrap:43:0)
at webpackJsonpCallback(file: app/webpack/bootstrap:30:0)
at anonymous(file:///app/bundle.js:2:61)
at evaluate([native code])
at moduleEvaluation
at
at asyncFunctionResume
at
at promiseReactionJob

farfromrefug, 2020-08-27:
You need to use @nativescript/core version "rc"

AvigarD611, 2020-08-27:
What do you mean by @nativescript/core version "rc"? I have installed @nativescript-community/ble. Then I've included import {Bluetooth} from '@nativescript-community/ble';
const bluetooth = new Bluetooth();

Then "bluetooth.connect....." . What do I need to do?

farfromrefug, 2020-08-27:
What version of nativescript are you using?

AvigarD611, 2020-08-27:
I am using last version with VueJS. I have used the old one library (which was before this one - nativescript-bluetooth 3.0.0@beta-10). With it I had no problem with connection/scanning/disconnecting/writing/reading. The only problem I had, was that OnNotify was never triggered.

So I saw your comment on #33 issue and installed this one, but this one fails on connect.

farfromrefug, 2020-08-27:
Please don't say latest ! Give the version number please! @nativescript-community/ble is for N 7.0 which is a breaking change and currently only in rc(or beta I should say)

AvigarD611, 2020-08-27:
Nativescript version is 6.8.0 and node is 6.14.6

farfromrefug, 2020-08-27:
As I said it is for N 7.x. If using 6.x stick with nativescript-bluetooth, beta 13 is out

AvigarD611, 2020-08-27:
is it right nativescript-bluetooth@3.0.0-beta13?

AvigarD611, 2020-08-27:
I just have tried beta-13 and it still doesn't work. Steps I am taking:
1) Scan
2) Connect
3) On button tap - startNotifying (it says that i am successfully subscribed in .then() )
4) On another button tap - write - value is written, but no notification comes

farfromrefug, 2020-08-28:
did you try with any ble test app? Are you sure your char is working ? i would go for an issue somewhere else

AvigarD611, 2020-08-28:
Yes, I did with 2 apps from AppStore. When I write, it receives notification with the right values from BLE.

farfromrefug, 2020-08-28:
Then we need a sample project to repoduce the issue

AvigarD611, 2020-08-28:
Here is the project. Its basically 1 page with few methods. Just npm install, after unzipping.

[nativescript-bluetooth.zip](https://github.com/nativescript-community/ble/files/5140596/nativescript-bluetooth.zip)

farfromrefug, 2020-08-28:
@AvigarD611 the issue is that i need your ble device, right?
Can you simulate it with [this](https://github.com/nativescript-community/ble/tree/master/ble_server_test)?

AvigarD611, 2020-08-28:
I can, but can you tell me how to use the files in the link? How I am supposed to build this "app". Also can you share skype/viber or something else where we can chat, so we dont spam here. Also it will be much faster to see answers

farfromrefug, 2020-08-28:
@AvigarD611 it is a node js app, no building just running. Let s chat on the nativescript slack, can benefit others.

AvigarD611, 2020-09-03:
Everything is working fine. The problem was on my side, because I was writing values as a string in hex, it should be array in dec. Thanks for support @farfromrefug !