nativescript-community/ble #191: How can I negate byte values when writing to a service?

Tyler-V, 2020-09-14:
The documentation of the bluetooth device I am working with says I need to negate the byte value of the third position of the byte array to mirror the second value.

```
public class GetInstantParaPkg {
    private byte[] buf;

    public GetInstantParaPkg() {
        buf = new byte[BTConstant.COMMON_PKG_LENGTH];
        buf[0] = (byte)0xAA;
        buf[1] = BTConstant.CMD_WORD_INSTANT_PARA;
        buf[2] = ~BTConstant.CMD_WORD_INSTANT_PARA;
        buf[buf.length-1] = CRCUtils.calCRC8(buf);
    }

    public byte[] getBuf() {
        return buf;
    }
}
```

How can I represent the above syntax through nativescript marshalling and what-not?

```
    writeWithoutResponse(service: Service, characteristic: Characteristic) {
        this.bluetooth
            .writeWithoutResponse({
                peripheralUUID: this.peripheral.UUID,
                serviceUUID: service.UUID,
                characteristicUUID: characteristic.UUID,
                value: ['0xAA', '0x17', '~0x17', '0x00', '0x00', '0x00', '0x00', '0xc6'],
            })
            .then(
                (result) => {
                    console.log(JSON.stringify(result));
                },
                (error) => {
                    console.log('Read error: ' + error);
                }
            );
    }
```

NathanaelA, 2020-09-14:
` value: ['0xAA', '0x17', '~0x17', '0x00', '0x00', '0x00', '0x00', '0xc6'],` 
should probably be:
` value: [0xAA, 0x17, ~0x17, 0, 0, 0, 0, 0xc6],`
You don't want to wrap them as a string...

However you might also need to make a manual byte array like so:
const buff = Array.create('byte',8);
then buff should be assigned the same way...
and passed into the `value:` -- auto marshalling might handle that, but if it doesn't -- use the `Array.create` method...

farfromrefug, 2020-09-15:
Indeed don't use strings. And in fact I would use a buffer that way it would also work on iOS

closed 2020-11-13 by farfromrefug