nativescript-community/ble #194: Android 10 - requestLocationPermission always rejected (Permission denied), even when granted

mst-marcernst, 2020-10-27:
Hi, thanks for your work - first of all!

I'm encountering an issue on Android 10 (OxygenOS 10.0.9). Calling 'requestLocationPermission' causes the permission dialog to open initially, but regardless of the user's choice, the Promise is always being rejected and returns "Permission denied".

If the user denies location permission, calling 'requestLocationPermission' again will display the popup a second time, offering the "Deny & don't ask again" option - however, granting permission either on the first or the second time will cause any subsequent attempts of calling 'requestLocationPermission' to be immediately rejected without the dialog being shown.

However, if the user grants permission when the permission dialog is being displayed (the first and only time), all following Bluetooth operations do work (scanning, connecting, etc.), despite the 'hasLocationPermission' promise returning false and 'requestLocationPermission' is being rejected without prompting the user.

`    private initialiseBluetooth(): Promise<boolean> {
        var _instance = this;

        return new Promise(function(resolve, reject) {
            // enable Bluetooth
            _instance._bluetooth.enable().then(
                enabled => {
                    //
                    console.log("Bluetooth enabled: " + enabled);
                    //
                    if(enabled) {
                        // check location permission
                        _instance._bluetooth.hasLocationPermission().then(
                            hasPermission => {
                                //
                                console.log("Location permission: " + hasPermission);
                                //

                                // HAS PERMISSION IS NEVER TRUE
                                if(hasPermission) {
                                    resolve(true);
                                }
                                // THUS WE ALWAYS END UP HERE
                                else {
                                    // request location permission
                                    _instance._bluetooth.requestLocationPermission().then(
                                        // THIS IS NEVER EXECUTED
                                        hasPermission => {
                                            //
                                            console.log("Permission granted: " + hasPermission);
                                            //
                                            resolve(hasPermission);
                                        },
                                        // INSTEAD, THE REQUEST IS ALWAYS REJECTED
                                        err => reject(new Error("Request location error: " + err))
                                    )
                                }
                            },
                            err => reject(new Error("Has permission error: " + err))
                        );
                    }
                    else resolve(false);
                },
                err => reject(new Error("Error enable Bluetooth: " + err))
            );
        });
    }
`

mst-marcernst, 2020-10-27:
Nevermind - I'm just stupid. Adding the permissions to the Android manifest does help :/

closed 2020-10-27 by mst-marcernst

matthew798, 2021-05-04:
@mst-marcernst  Your not stupid. Don't say that. This solved my problem, so thanks.