nativescript-community/ble #195: Bluetooth causes NativeScript app to crash on ios 14.1

jthompson-ceres, 2020-11-02:
After running:
```bash
>> ns run ios
```

When calling bluetooth is enable in the app:
```js
this.bluetooth.isBluetoothEnabled().then(
          function(enabled) {
            console.log("Enabled: " + enabled);
          }
);
```

NativeScript app crashes with error:
"Running performTccCheck CBManager tccAvail 1, tccRequired 1
TCC required
This app has crashed because it attempted to access privacy-sensitive data without a usage description.  The app's Info.plist must contain an NSBluetoothAlwaysUsageDescription key with a string value explaining to the user how the app uses this data."

I've already added (with description) in info.plist:
Bluetooth Always Usage Description

Still did not fix the issue....
Any suggestions?

farfromrefug, 2020-11-02:
@jthompson-ceres look at the doc there are new descriptions you need to add (don't have my computer around)

JoshuaBThompson, 2020-11-03:
@farfromrefug (this is jthompson-ceres)
Which docs are you referring too? Apple?

JoshuaBThompson, 2020-11-03:
@farfromrefug ahhh found the issue!
I was adding the key to the wrong info.plist
Adding it to App_Resources/iOS/Info.plist does the trick.
According to the NativeScript docs here:
https://docs.nativescript.org/tooling/publishing/publishing-ios-apps#example-1-how-to-customise-infoplist__

NathanWalker, 2020-11-05:
There's another issue here I found in another user's app:
https://github.com/NativeScript/NativeScript/issues/8999#issuecomment-722111181

farfromrefug, 2020-11-05:
@NathanWalker thanks good.find but weird one :)

farfromrefug, 2020-11-05:
published 3.0.7 with build.xcconfig

closed 2020-11-13 by farfromrefug