nativescript-community/ble #196: DiscoverCharacteristics not returning promise iOS

jthottel, 2020-11-06:
The device I'm working against has a service 0000ffe0-0000-1000-8000-00805f9b34fb which is discovered just fine with discoverServices. When I call discoverCharacteristics with that ID the promise never gets returned. I did a little debugging and for some reason in the args for discoverCharacteristics in bluetooth.ios.js the id is coming through as ffe0 but everything seems fine until we get to this if statement.

**if (UUID === pUUID && sUUID === args.serviceUUID)**

Here the sUUID does not equal args.serviceUUID and this is where the call will hang at. 

As far as I can tell I have 2 issues here.
1. The args isn't coming through properly. 
2. The if statement above should return the promise as a reject if it is not met.

farfromrefug, 2020-11-06:
@jthottel thanks. Would you l'île to make a PR?

jthottel, 2020-11-06:
I can make a PR to fix the if statement issue but I still can't explain why on this particular service the arg doesn't get passed correctly.

farfromrefug, 2020-11-07:
@jthottel about the id is coming through as ffe0 i think it is a standard behavior for know services/chars in that "namespace". Can you try using ffe0 as serviceUUID?

farfromrefug, 2020-12-10:
@jthottel i will close this. Let me know if you still have an issue