nativescript-community/ble #197: "bluetooth-central" error

MarkKewley, 2020-11-10:
Hello,

This is probably just me not having everything configured correctly so hopefully it's a simple fix. For reference I'm using the Nativescript playground app/preview app and am not running a simulator. My IOS version is 14.1.

I'm seeing the following error whenever I try checking if bluetooth is enabled or scanning (frankly any useful operation).

```
    try {
      const enabled = await bluetooth.isBluetoothEnabled();
      console.info('Bluetooth enabled', enabled);
    } catch (e) {
      console.error('Error checking if Bluetooth was enabled', e);
    }
```

Logs:
```
Error checking if Bluetooth was enabled Error: State restoration of CBCentralManager is only allowed for applications that have specified the "bluetooth-central" background mode
```

I checked some of the closed threads around here as well as read some of the [apple docs](https://developer.apple.com/library/archive/documentation/NetworkingInternetWeb/Conceptual/CoreBluetooth_concepts/CoreBluetoothBackgroundProcessingForIOSApps/PerformingTasksWhileYourAppIsInTheBackground.html) and the common solution was to enable bluetooth-central (duh!).

I've added the following to the `Info.plist` yet I'm still receiving the error:

```
	<key>UIBackgroundModes</key>
	<array>
		<string>bluetooth-central</string>
	</array>
	<key>NSBluetoothAlwaysUsageDescription</key>
	<true/>
```

I'm very new to NativeScript and it's been over 4 years since I've written any sort of Native Bluetooth app using Swift. 

Here is my package.json for reference:
```
  "dependencies": {
    "@nativescript-community/ble": "^3.0.8",
    "@nativescript/core": "^7.0.3",
    "@nativescript/theme": "^2.3.3",
    "@vue/devtools": "^5.3.3",
    "nativescript-socketio": "^3.3.1",
    "nativescript-toasty": "^3.0.0-alpha.2",
    "nativescript-vue": "^2.8.1",
    "nativescript-vue-devtools": "^1.4.0",
    "vuex": "^3.5.1"
  },
  "devDependencies": {
    "@babel/core": "^7.11.6",
    "@babel/preset-env": "^7.11.5",
    "@nativescript/ios": "7.0.5",
    "@nativescript/types": "^7.0.0",
    "@nativescript/webpack": "^3.0.4",
    "@types/node": "^14.6.2",
    "babel-loader": "^8.1.0",
    "nativescript-vue-template-compiler": "^2.8.1",
    "nativescript-worker-loader": "~0.12.1",
    "sass": "^1.26.10",
    "typescript": "^3.9.7",
    "vue": "^2.6.12",
    "vue-loader": "^15.9.3"
  }
```

farfromrefug, 2020-11-13:
Can't really help without seeing your app. Maybe you should compare to the demo app. Especially the info.plist

mbaush77, 2020-11-13:
Try this: 
```
<key>NSBluetoothAlwaysUsageDescription</key>
<string>To find your computer in proximity</string>
<key>UIBackgroundModes</key>
<array>
    <string>bluetooth-central</string>
</array>
```

MarkKewley, 2020-11-14:
Thanks for getting back to me!

@farfromrefug 
Apologies, all it is is a raw VueJS NativeScript application following the install guide [here](https://nativescript-vue.org/en/docs/getting-started/quick-start) with the added ble library.

So very bare-bones. I've only changed the App.vue mounted

```
  async mounted() {
    const bluetooth = new Bluetooth();

    console.info('Checking if BLE is enabled');
    try {
      const enabled = await bluetooth.isBluetoothEnabled();
      console.info('Bluetooth enabled', enabled);
    } catch (e) {
      console.info('Error checking if Bluetooth was enabled', e);
    }
  },
```

I compared (copied I should say) the `Info.plist` from the demo but was still receiving the error

@mbaush77 Thanks for the tip! I tried that out but was still getting the aforementioned error

farfromrefug, 2020-11-14:
@MarkKewley please share the project

MarkKewley, 2020-11-14:
@farfromrefug 
Sure thing! Here you go:

[spotify-lights.zip](https://github.com/nativescript-community/ble/files/5541549/spotify-lights.zip)

MarkKewley, 2020-11-27:
@farfromrefug just checking in, have you been able to see any issues with my setup?

MarkKewley, 2021-02-03:
Closing as I'm fairly certain this is a result of running on simulator vs real device

closed 2021-02-03 by MarkKewley