nativescript-community/ble #198: startScanning won't find any devices when app is in background

mbaush77, 2020-11-13:
I have a worker service that continues scanning for bluetooth devices when the app is no longer in foreground. I still get console logs from the worker when I put the app in background but the startScanning() method no longer returns bluetooth devices. (they are returned fine when the app is in foreground)

I'm assuming this is something handled at the OS level that needs to be configured or something like in the Info.plist keys maybe?

Thanks!

farfromrefug, 2020-11-14:
@mbaush77 never done that. You have to look at the native APIs docs

LaKing, 2021-04-17:
I see the same behaviour. ...

@mbaush77 could you overcome it?

farfromrefug, 2021-04-17:
@LaKing you need to seaech on the native side but it might not be possible