nativescript-community/ble #201: iOS: Enable bluetooth works the first time but not second

schnapzz, 2020-12-08:
**Project specs**
node: v14 LTS
ns-cli: latest - 7.0.11
ns-core: latest - 7.0.13
ns-ios: latest -  7.0.6
ns-community/ble: latest - 3.0.12

**Issue**
After upgrading my project to {N} 7 I have an issue with the method isBluetoothEnabled on iOS. First time the method is call it correctly create a dialog window asking for bluetooth permission. But subsequent calls defaults it to false and does not trigger a new dialog.
Here's a bit of my code,

```
bluetooth.isBluetoothEnabled().then(
  function(enabled) {
    console.log("Enabled? " + enabled);
    bluetooth.startScanning(...)
  }
);
```
Before upgrading to {N} 7 from {N} 6 and used the nativescript-bluetooth 1.3.1, this worked like a charm, same exact code. But now enable is only true for the first time and no options is given to the user. 

Even if I ignore calling isBluetoothEnabled, the following exception occours when scanning:
` BT Scanning caught exception: Error: bluetooth_not_enabled`

Anybody got an idea why the bluetooth permission isn't properly handled or have suggestions to what I can try? Cheers in advance.

farfromrefug, 2020-12-10:
@schnapzz just verifed and it seems to be working fine. You can test the demo app. You ll see that (normally) clicking on the enabled button always returns the correct state.

schnapzz, 2020-12-16:
Thanks for the answer and sorry for the long reply @farfromrefug. I haven't had the time to figure if it's then caused by some of my code. Will return when I've had the opportunity to look into it :)

schnapzz, 2021-01-07:
@farfromrefug I've not investigated and it was an issue in my end that was hiding itself fairly well. Appologise wasting some of your time, but thanks for the reply :-) 

I'll close the issue as well.

closed 2021-01-07 by schnapzz

Atomico001, 2021-05-04:
I have this problem too, how you fix it?