nativescript-community/ble #205: Is there something which I can't do with this module?

Cosbgn, 2021-02-04:
Hello,
I'm developing an application which uses BLE to connect to a device. 
I'm getting some quotes from Agencies and one is telling me that I can't use nativescript because bluetooth support on cross platform is limited and I should instead develop 3 different native apps (web + ios + android).

Is this true? 
Is there something which is really impossible with nativescript and requires 3 native apps?
Thank you for the answers

farfromrefug, 2021-02-04:
@Cosbgn this is totally wrong! cross platform does not have any implication on ble support. Our plugin ble support is already really good and used in multiple production apps.
You can contact me or other N devs if you want more details or even quotes ;)
We all are on the nativescript community slack

Cosbgn, 2021-02-04:
Thank @farfromrefug that's what I thought. I've passed the feedback to the team and they might reach out on Slack!

closed 2021-02-04 by Cosbgn