nativescript-community/ble #206: ios finds most devices only once

ray007, 2021-02-15:
I recently changed my code from 5 seconds scanning and manual refresh to continuous scanning and automatic removal of devices that haven't been seen for more than 5 seconds from the list.

On android this works fine, on iOS this fills my list with all nearby devices matching the filter (between 5 and 9), but then I only seem to receive update for exactly 1 device. So after 6-7 seconds I have only one remaining entry in the list.

After stopping the scan and restarting it again after a while, the same happens again. It's even always the same device which remains to get updates, and it's not the closest one.

All this with `@nativescript-community/ble` version `3.0.13`, `@nativescript/core` version `7.2.1` and `@nativescript/ios` version `7.1.1` or `7.2.0`.

farfromrefug, 2021-02-15:
this lib won't send you scan updates for devicea already seen. this is intended.
ble discovery is a big power drain, this lib is not intended to be continuously be discovering.
that being said PRs are welcome. you could either add an optional continuous mode where you would get every device scan update. or you could add a method to "forget" a discovered device so that you would get a new device discovery event

ray007, 2021-02-16:
Without finding a sensor more than once you cannot calculate an average to guess which one is nearest.
Why do I get continuous updates for 1 sensor on iOS then?
And on Android the behavior is as expected, I get continuous scan results for all sensors nearby.

Also: not passing on messages for devices you've seen once does not reduce the power drain.

ray007, 2021-02-16:
I tried to add an option to get duplicate results as you suggested with 
```javascript
this.centralManager.scanForPeripheralsWithServicesOptions(services, {"CBCentralManagerScanOptionAllowDuplicatesKey": "true"});
```
but that did not change anything.
Did I do something wrong when trying to supply the option?

farfromrefug, 2021-02-16:
@ray007 Then i must not understand what you want.
ON android we only trigger discover if not already discovered https://github.com/nativescript-community/ble/blob/master/src/bluetooth.android.ts#L512
Indeed on iOS we always trigger.

ray007, 2021-02-16:
Very strange, this does not match my observation.
I have
```var BTI = getBluetoothInstance();```
and later in my scanning function:
```javascript
BTI.startScanning({
	filters: [{serviceUUID: "78a....."}],
	onDiscovered: dev => {
		... do some stuff here ...
	}
});
```
The `onDiscovered` callback is called repeatedly for all nearby devices on android, but not on iOS.

I tried to find the caller with the debugger, but the source mapping data seems to be wrong/corrupt, so no luck there.

farfromrefug, 2021-02-16:
@ray007 sourcemap is good, i use it here. The issue is on the "client" side which must be wrongly translating the source map.
as for your issue i cant help you if i cant reproduce it here. The best would be for you to debug through the js side of this plugin and see where it fails.

ray007, 2021-02-16:
I have no idea what going wrong here with webpack, both `@nativescript/webpack` version 3 and 4 do not show me the correct lines in the debugger when going back the call stack of `onDiscovered`.

And I am confused about another point:
You point me to the `onLeScan` function passed to the `super` contructor of `class LeScanCallbackImpl` (around line 512), while my `onDiscovered` callback seems to be called from the `ScanCallBackImpl.onScanResult` method (see https://github.com/nativescript-community/ble/blame/master/src/bluetooth.android.ts#L639)
The 2 methods looks quite similar, but the `if (!stateObject)` closes much earlier in the latter.

farfromrefug, 2021-02-16:
you are correct I pointed you to the preview lollipop version. but the other one does the same (not trigger if already discovered)

ray007, 2021-02-17:
Did you do anything special to make debugging this work for you? I also see devtools are displaying the `.js` files from the `ble` module distribution, but the tab title suggests it wants to display the `.ts` source files.

As for my apple-question above, [apple documentation for the scanning function](https://developer.apple.com/documentation/corebluetooth/cbcentralmanager/1518986-scanforperipheralswithservices) shows an [options](https://developer.apple.com/documentation/corebluetooth/cbcentralmanager/peripheral_scanning_options) parameter.
And [CBCentralManagerScanOptionAllowDuplicatesKey](https://developer.apple.com/documentation/corebluetooth/cbcentralmanagerscanoptionallowduplicateskey) looked like the solution to my problem.

But either I misread the documentation or my try to supply the parameter as key in an object is wrong.

And while I understand your argument about the power drain, I'd have thought the solution to that would be to stop scanning, not to skip calling the discover-callback.
The idea here is that the user can walk through the house with the "Discover" tab of the application open, and choosing a sensor opens another tab and stops scanning.

farfromrefug, 2021-02-17:
@ray007 your use case is very different from the simple connect a ble device use case. In most cases you dont need to get continuous scan updates. But i understand what you want.
Which is why i told you i would gladly accept PR to add that feature.

Now about debugging i dont use any "debugger" simply the console so cant help there.

ray007, 2021-02-17:
I'd be happy/willing to make a PR for this once I know how to.
For Android this seems to be rather straight forward, but I cannot do so for iOS since I do not yet know what I would have to change there. Any hint maybe if I did something wrong with setting the scanning options?

Which brings us back to debugging:
Looking at the `.js.map` files from installation with npm, they all have a property `"sourceRoot": "../src/"`.
And the `src` directory is not published to npm.

farfromrefug, 2021-02-17:
@ray007 first we should move that discussion to slack or even better to discord if you are there.
sourcemap: does not make a difference really. it is normal for the path. The thing is to get the line number right.

ray007, 2021-02-17:
I don't have a slack-account, but getting on discord should not be a problem. Got an invite for me or should I start something?

Sourcemap: it's definitely the `.js.map` files that are the problem. If I remove those before the build, going back the call stack in the debugger brings me to the correct place in the js files.

farfromrefug, 2021-02-17:
@ray007 https://discord.gg/6rUSWKCu

sebj54, 2022-02-18:
Hi @farfromrefug, sorry to dig up this old issue but I'm in the same case than @ray007: I get multiple scan results for the same device on Android and only one on iOS. 

I have a solution to make it report duplicates also on iOS:

This line: https://github.com/nativescript-community/ble/blob/a81ffcffdb6121559aaa3ad0cb5046845635410c/src/bluetooth.ios.ts#L827

becomes:

```ts
const options = new NSMutableDictionary([true], [CBCentralManagerScanOptionAllowDuplicatesKey]);
this.centralManager.scanForPeripheralsWithServicesOptions(services as any, options);
```

I wanted to add a `allowDuplicates` argument to the `startScanning` method. But when I'm reading your comments saying that you don't have duplicates also on Android, it makes me doubt. Arre you sure that this is the default behavior on Android? I try to search this on Google but did not find anything. I ended up reading code in the BLE Cordova plugin and they implement it logically to avoid duplicates reported several times: https://github.com/don/cordova-plugin-ble-central/blob/a9259ba6d7305890bec3985b16d3928ac6f4c6f6/src/android/BLECentralPlugin.java#L868-L871

I can make a PR if you think such a system would be good for this plugin.

sebj54, 2022-02-18:
Ok great! PR should be ready shortly.

farfromrefug, 2022-02-18:
@sebj54 thanks a lot!

closed 2022-02-18 by farfromrefug

ray007, 2022-02-21:
Seems to work fine, thanks a lot!

PR #241 (2022-02-18): feat: `avoidDuplicates` option on `startScanning` method
I finally chose to name the parameter `avoidDuplicates` instead of `allowDuplicates` so it is `false` by default. It simplifies the code as we don't have to set a default value if the parameter is `undefined`. Plus, it makes all boolean parameters of `startScanning` false by default.

Tested on both Android & iOS with and without `avoidDuplicates`.

Closes #206