nativescript-community/ble #207: Error with version 3.0.14

senner008, 2021-02-16:
Hi :)

When I run 
.startNotifying()
I get: 

`Uncaught (in promise): [BluetoothError]:service_not_found, undefined, undefined,`

With latest version 3.0.14 on Android 9

farfromrefug, 2021-02-16:
@senner008 you are trying to access a service which does not exist.

senner008, 2021-02-16:
Going back to 3.0.13 and the error goes away

farfromrefug, 2021-02-16:
@senner008 OK I see 
can you share the code of where it fails? can't help.without the code

radilr1, 2021-02-17:
I receive the same error on Android 10

farfromrefug, 2021-02-17:
@radilr1 @senner008 ok i might know what s going on.
In our connect call, could you add `autoDiscoverAll:true`  to see if it fixes it.
I might see the regresssion if it works

sebj54, 2021-02-17:
I had the same problem after upgrading to version 3.0.14. Downgrading fixes the problem.

I tested your suggestion (`autoDiscoverAll: true`) when connecting (with version 3.0.14 of course) and the error is gone now!

farfromrefug, 2021-02-17:
@sebj54 awesome! i ll wait on one more feedback before i publish the fix

senner008, 2021-02-18:
`autoDiscoverAll: true` 
fixed it on my end as well

farfromrefug, 2021-02-18:
@senner008 @radilr1 @sebj54 3.0.15 published with the fix! thanks all for your help

closed 2021-02-18 by farfromrefug

sebj54, 2021-02-18:
Thank you @farfromrefug for the quick fix!