nativescript-community/ble #209: error reading characteristic on android

ray007, 2021-03-01:
After my most recent node modules update, reading a characteristic stopped working on android, everything still works fine on ios.
Testing older versions the problem popped up with version `3.0.14`, version `3.0.13` still works fine.

The error returned is `[BluetoothError]:service_not_found, undefined, undefined, ...`, but the service I'm trying to read a characteristic from is the same one I'm using to filter results.

farfromrefug, 2021-03-01:
fixed in 3.0.15

closed 2021-03-01 by farfromrefug

ray007, 2021-03-02:
That is true for current android, but it is not for `Android 9` on my `Samsung Galaxy S8+` test device.

farfromrefug, 2021-03-02:
then we need a repoducaable project and a lot more details on what you are seeing

ray007, 2021-03-03:
I'm using a non-standard serviceUUID with non-standard characteristic IDs.

Note: adding `"inlineSources": true` to the `compilerOptions` in `tsconfig.json` also let's people just installing the package with npm step back into your code during debugging.

I can catch the error in `bluetooth.android.ts` line `2635`.

```json
Bluetooth error: {
	"name": "BluetoothError",
	"method": "readCharacteristic",
	"message": "error_function_call",
	"status": 133,
	"arguments": {
		"peripheralUUID": "...", "serviceUUID": "...", "characteristicUUID": "..."
	}
}
```

The `err.stack` is unfortunately less helpful since it just points into my `vendor.js`:
```
BluetoothError: error_function_call
    at Object.onCharacteristicRead (file:///data/data/com.app.id/files/app/vendor.js:2254:23)
    at file:///data/data/com.app.id/files/app/vendor.js:1186:13
    at Array.forEach (<anonymous>)
    at Object.BluetoothGattCallbackImpl.onCharacteristicRead (file:///data/data/com.app.id/files/app/vendor.js:1184:25)
```

Now things get funky.
I've seen the read-error at least 20 times, and at least 5 of those during writing this post.

`BluetoothGattCallbackImpl#onCharacteristicRead` is not a place where I could set a breakpoint in the debugger, so I set one in function `onCharacteristicRead` on line `1827`.

Since stopping at this breakpoint and stepping to the end of the function, the problem is gone.

So it seems the problem has vanished for now, but I have no idea idea what caused or fixed it.

Thanks for your help, I'll post here again if/when the problem pops up again.

farfromrefug, 2021-03-03:
@ray007 this is not the same error! now you are showing read char error s with code 133(you can look it up on the web) while before it was service not found (the error fixed on 3.0.15)