nativescript-community/ble #210: Could not recognise ATT errors returned from GATT server

matteonatale, 2021-03-02:
In my typescript code I couldn't catch some errors returned from my GATT server implementation (e.g. the `StatusUnexpectedError` constant defined in BLE spec). The following code executes without failures but on server logs I can see that the request is actually returning an error.

```ts
const packetToSend = new Uint8Array(...)

await this.bluetooth.write({
  peripheralUUID: this.peripheral.UUID,
  serviceUUID: serviceUuid,
  characteristicUUID: characteristicUuid,
  encoding: 'ASCII',
  value: packetToSend
})
```

Has anyone experienced the same problem? Thanks a lot! :D

farfromrefug, 2021-03-03:
you don't see any error on iOS nor Android ?

matteonatale, 2021-03-03:
Sadly no. I've also tried to return an error intentionally from the GATT server but nothing happens on the nativescript side, it just resolve without failures

farfromrefug, 2021-03-03:
@matteonatale so you need to "debug" it a bit more. i think android is your best bet to debug it as you can use logcat to see more ble messages. maybe we are missing a listener to catch that error
BTW this is a write with response ?

matteonatale, 2021-03-03:
Yes, it's a write with response

farfromrefug, 2021-03-03:
OK I looked at the code again and it should catch the non success status. are you sure your server "replies" with the error ?

matteonatale, 2021-03-03:
Yeah, I'm sure. However I'm going to try a little bit more, I'll let you know. 
Just to be certain: in order to catch ATT errors can I use the try/catch statement or I should get it in another way?

farfromrefug, 2021-03-03:
the write function is a promise so you need to catch the promise error

matteonatale, 2021-03-05:
I implemented a new GATT server using another package (the server is written in Go) in order to get the error from the promise, but unfortunately I still can't catch it. I gave up the idea to retreive the error from the write method because i thought that for my use it's better to start notifying and listening for an eventually error message notified by the server. However thanks for the reply! :D

closed 2021-03-05 by farfromrefug