nativescript-community/ble #213: List of paired devices

arturikoX, 2021-03-24:
Hi,

Can I know the paired devices my phone currently hold?
The problems I'm facing is that the "Disconnect" method keeps connections to previously devices,
So when I make another "scan" devices - I can't see them..

Versions:
    "nativescript-bluetooth": "^3.0.0"
    "tns-ios": {
      "version": "6.5.3"
    },
    "tns-android": {
      "version": "6.5.3"
    }

Thanks

farfromrefug, 2021-03-24:
@arturikoX there is an undocumented `isConnected` method using the sames params as `connect`

arturikoX, 2021-03-24:
Hi,
Thank you for you reply,
I still have issue with Android and disconnecting ( On ios the disconnect works perfectly),
is the any new approach for disconnecting on newer versions of this plugin?

farfromrefug, 2021-03-24:
@arturikoX I don't understand your issue. please explain in details and with sample code if possibke

arturikoX, 2021-03-24:
@farfromrefug ,
After disconnecting from a ble device and getting:
JS: [nativescript-bluetooth] gattDisconnect ---- Closing GATT client 9C:A5:25:AE:E3:87 9C:A5:25:AE:E3:87 2

the device is still connecting in some way,
I still receiving "onNotify" events, I think this issue https://github.com/nativescript-community/ble/issues/70 relevant to my issue..

farfromrefug, 2021-03-24:
@arturikoX i have never see n that issue. if you van reproduce it by emulating a device with noble then i can fix it. 
look it `ble_server_test`