nativescript-community/ble #215: Connect causes Fatal JavaScript exception on iOS. Works fine on Android

jpmartinsson, 2021-04-15:
After upgrade from old NativeScript-bluetooth to @nativescript-community/ble I cannot connect to device from iOS (14). Scanning works find and the device is found, but in the connect-call I get. I realise this has something to do with our project setup, but can anyone point us in the right direction?:

```
21-04-15 15:48:32.060135+0200 triplyui[2392:639104] ***** Fatal JavaScript exception - application has been terminated. *****
2021-04-15 15:48:32.060446+0200 triplyui[2392:639104] Native stack trace:
2021-04-15 15:48:32.060794+0200 triplyui[2392:639104] 1   0x1059e2348 NativeScript::reportFatalErrorBeforeShutdown(JSC::ExecState*, JSC::Exception*, bool)
1   0x1059e2348 NativeScript::reportFatalErrorBeforeShutdown(JSC::ExecState*, JSC::Exception*, bool)
2021-04-15 15:48:32.061231+0200 triplyui[2392:639104] 2   0x105a1c434 NativeScript::FFICallback<NativeScript::ObjCMethodCallback>::ffiClosureCallback(ffi_cif*, void*, void**, void*)
2   0x105a1c434 NativeScript::FFICallback<NativeScript::ObjCMethodCallback>::ffiClosureCallback(ffi_cif*, void*, void**, void*)
2021-04-15 15:48:32.061542+0200 triplyui[2392:639104] 3   0x106587388 ffi_closure_SYSV_inner
3   0x106587388 ffi_closure_SYSV_inner
2021-04-15 15:48:32.061822+0200 triplyui[2392:639104] 4   0x1065881b4 .Ldo_closure
4   0x1065881b4 .Ldo_closure
2021-04-15 15:48:32.061971+0200 triplyui[2392:639104] 5   0x1b28f8708 <redacted>
5   0x1b28f8708 <redacted>
2021-04-15 15:48:32.062082+0200 triplyui[2392:639104] 6   0x1b28f5354 <redacted>
6   0x1b28f5354 <redacted>
2021-04-15 15:48:32.062181+0200 triplyui[2392:639104] 7   0x1b28ea584 <redacted>
7   0x1b28ea584 <redacted>
2021-04-15 15:48:32.062288+0200 triplyui[2392:639104] 8   0x1b2911a2c <redacted>
8   0x1b2911a2c <redacted>
2021-04-15 15:48:32.062391+0200 triplyui[2392:639104] 9   0x1b2905754 <redacted>
9   0x1b2905754 <redacted>
2021-04-15 15:48:32.062494+0200 triplyui[2392:639104] 10  0x106b5fbcc _dispatch_call_block_and_release
10  0x106b5fbcc _dispatch_call_block_and_release
2021-04-15 15:48:32.062596+0200 triplyui[2392:639104] 11  0x106b616c0 _dispatch_client_callout
11  0x106b616c0 _dispatch_client_callout
2021-04-15 15:48:32.062881+0200 triplyui[2392:639104] 12  0x106b69354 _dispatch_lane_serial_drain
12  0x106b69354 _dispatch_lane_serial_drain
2021-04-15 15:48:32.062995+0200 triplyui[2392:639104] 13  0x106b6a0f4 _dispatch_lane_invoke
13  0x106b6a0f4 _dispatch_lane_invoke
2021-04-15 15:48:32.063095+0200 triplyui[2392:639104] 14  0x106b70d7c _dispatch_main_queue_callback_4CF
14  0x106b70d7c _dispatch_main_queue_callback_4CF
2021-04-15 15:48:32.063259+0200 triplyui[2392:639104] 15  0x198c0d11c <redacted>
15  0x198c0d11c <redacted>
2021-04-15 15:48:32.063437+0200 triplyui[2392:639104] 16  0x198c07120 <redacted>
16  0x198c07120 <redacted>
2021-04-15 15:48:32.063730+0200 triplyui[2392:639104] 17  0x198c0621c CFRunLoopRunSpecific
17  0x198c0621c CFRunLoopRunSpecific
2021-04-15 15:48:32.063989+0200 triplyui[2392:639104] 18  0x1b07d0784 GSEventRunModal
18  0x1b07d0784 GSEventRunModal
2021-04-15 15:48:32.064670+0200 triplyui[2392:639104] 19  0x19b646ee8 <redacted>
19  0x19b646ee8 <redacted>
2021-04-15 15:48:32.065469+0200 triplyui[2392:639104] 20  0x19b64c75c UIApplicationMain
20  0x19b64c75c UIApplicationMain
2021-04-15 15:48:32.065719+0200 triplyui[2392:639104] 21  0x106588044 ffi_call_SYSV
21  0x106588044 ffi_call_SYSV
2021-04-15 15:48:32.065924+0200 triplyui[2392:639104] 22  0x106586ae8 ffi_call_int
22  0x106586ae8 ffi_call_int
2021-04-15 15:48:32.066137+0200 triplyui[2392:639104] 23  0x106586648 ffi_call
23  0x106586648 ffi_call
2021-04-15 15:48:32.066329+0200 triplyui[2392:639104] 24  0x1059a53d8 NativeScript::FunctionWrapper::call(JSC::ExecState*)
24  0x1059a53d8 NativeScript::FunctionWrapper::call(JSC::ExecState*)
2021-04-15 15:48:32.066559+0200 triplyui[2392:639104] 25  0x106553b10 llint_entry
25  0x106553b10 llint_entry
2021-04-15 15:48:32.066808+0200 triplyui[2392:639104] 26  0x106550f28 llint_entry
26  0x106550f28 llint_entry
2021-04-15 15:48:32.067026+0200 triplyui[2392:639104] 27  0x106550f28 llint_entry
27  0x106550f28 llint_entry
2021-04-15 15:48:32.067231+0200 triplyui[2392:639104] 28  0x106550f28 llint_entry
28  0x106550f28 llint_entry
2021-04-15 15:48:32.067431+0200 triplyui[2392:639104] 29  0x106550f28 llint_entry
29  0x106550f28 llint_entry
2021-04-15 15:48:32.067620+0200 triplyui[2392:639104] 30  0x106550f28 llint_entry
30  0x106550f28 llint_entry
2021-04-15 15:48:32.067815+0200 triplyui[2392:639104] 31  0x106550f28 llint_entry
31  0x106550f28 llint_entry
2021-04-15 15:48:32.067966+0200 triplyui[2392:639104] JavaScript stack trace:
2021-04-15 15:48:32.068155+0200 triplyui[2392:639104] peripheralDidDiscoverServices(file:///app/vendor.js:86221:134)
at file:///app/vendor.js:85076:48
[... 130 lines cut ...]
    "@angular/compiler-cli": "~8.2.0",
    "@angular/language-service": "^8.2.6",
    "@nativescript/schematics": "~0.7.0",
    "@nativescript/tslint-rules": "~0.0.2",
    "@types/jasmine": "~2.8.3",
    "@types/jasminewd2": "~2.0.2",
    "@types/node": "^10.17.21",
    "codelyzer": "^5.2.2",
    "jasmine-core": "~2.8.0",
    "jasmine-spec-reporter": "~4.2.1",
    "karma": "^4.3.0",
    "karma-chrome-launcher": "~2.2.0",
    "karma-coverage-istanbul-reporter": "^1.2.1",
    "karma-jasmine": "~1.1.0",
    "karma-jasmine-html-reporter": "^0.2.2",
    "nativescript-dev-webpack": "^1.5.1",
    "protractor": "^6.0.0",
    "style-loader": "^1.2.1",
    "tns-android": "6.5.3",
    "tns-ios": "6.5.4",
    "tns-platform-declarations": "^6.5.1",
    "ts-node": "~4.1.0",
    "tslint": "~5.9.1",
    "typescript": "~3.5.3"
```

jpmartinsson, 2021-04-15:
I can also add that the Device is actually connected (When looking at Bluetooth Connections on the phone)

jpmartinsson, 2021-04-15:
I have now updated my project to NS7 and Angular 10. I was hoping this could miraculously solve the issue, but unfortunately the error message remains exactly the same:

```
"dependencies": {
    "@angular/animations": "10.1.0",
    "@angular/common": "10.1.0",
    "@angular/compiler": "10.1.0",
    "@angular/core": "10.1.0",
    "@angular/forms": "10.1.0",
    "@angular/platform-browser": "10.1.0",
    "@angular/platform-browser-dynamic": "10.1.0",
    "@angular/router": "10.1.0",
    "@dsvishchov/nativescript-ngx-shadow": "^7.0.3",
    "@nativescript-community/ble": "^3.0.16",
    "@nativescript/angular": "10.1.2",
    "@nativescript/core": "7.3.0",
    "@nativescript/datetimepicker": "~2.0.0",
    "@nativescript/theme": "~2.2.1",
    "@nativescript/webpack": "3.0.0",
    "@proplugins/nativescript-localstorage": "^2.3.1",
    "nativescript-directions": "^1.3.1",
    "nativescript-geolocation": "^5.1.0",
    "nativescript-permissions": "^1.3.9",
    "nativescript-ui-gauge": "~7.0.2",
    "reflect-metadata": "~0.1.12",
    "rxjs": "^6.4.0",
    "rxjs-compat": "^6.0.0-rc.0",
    "tslib": "^1.11.2",
    "zone.js": "~0.10.3"
  },
  "devDependencies": {
    "@angular-devkit/build-angular": "0.1000.8",
    "@angular/cli": "^10.0.0",
    "@angular/compiler-cli": "10.1.0",
    "@angular/language-service": "^8.2.6",
    "@nativescript/schematics": "~0.7.0",
    "@nativescript/tslint-rules": "~0.0.2",
    "@types/jasmine": "~2.8.3",
    "@types/jasminewd2": "~2.0.2",
    "@types/node": "^10.17.21",
    "codelyzer": "^5.2.2",
    "jasmine-core": "~2.8.0",
    "jasmine-spec-reporter": "~4.2.1",
    "karma": "^4.3.0",
    "karma-chrome-launcher": "~2.2.0",
    "karma-coverage-istanbul-reporter": "^1.2.1",
    "karma-jasmine": "~1.1.0",
    "karma-jasmine-html-reporter": "^0.2.2",
    "protractor": "^6.0.0",
    "style-loader": "^1.2.1",
    "tns-android": "6.5.3",
    "tns-ios": "6.5.4",
    "ts-node": "~4.1.0",
    "tslint": "^5.16.0",
    "typescript": "3.9.7",
    "@nativescript/types": "7.0.0"
  }
```

farfromrefug, 2021-04-16:
@jpmartinsson You have some weird setup issue or something. Your issue is because an import is not found from N
```
2021-04-15 15:48:32.068435+0200 triplyui[2392:639104] file:///app/vendor.js:86221:134: JS ERROR TypeError: undefined is not an object (evaluating '_nativescript_core_utils_native_helper__WEBPACK_IMPORTED_MODULE_2__["iOSNativeHelper"].collections')
```
Though this file exists https://github.com/NativeScript/NativeScript/blob/master/packages/core/utils/native-helper.ios.ts.
You need to try and find out why it is not "seen"

jpmartinsson, 2021-04-16:
Thanks! Can you see any connection between this error and the connect function.  I thought if it was a setup issue it would affect more things and not just the Connect call. The application dies exactly when the connection is completed, but no Promise has been returned. Where would you look for setup issues? The native-helper-files are in my node_modules (see attached image)
[2021-04-16_11-35-31.pdf](https://github.com/nativescript-community/ble/files/6324025/2021-04-16_11-35-31.pdf)

farfromrefug, 2021-04-16:
@jpmartinsson ok you need to check your webpack config maybe to make sure it manage to "choose" native-helper.ios.js or native-helper.android.js
I am quite lost here without being able to reproduce here

dimitriospafos, 2021-10-19:
@jpmartinsson  did you ever solve this issue? I'm getting the same error on Ios. Any help is appreciated. Thanks

farfromrefug, 2021-10-19:
@dimitriospafos the latest version requires N 8.x