nativescript-community/ble #216: Unable to create application

Mahtaran, 2021-04-15:
After doing a really basic install (just the hello world TypeScript project, and running `ns plugin add @nativescript-community/ble`), I get some weird error when I launch my application on the android emulator. This does not happen when I remove the BLE package. I am using NativeScript 8.0.0, and am quite new to this ecosystem, so please let me know if I omitted some of the necessary information.

The stacktrace:
```
An uncaught Exception occurred on "main" thread.
Unable to create application com.tns.NativeScriptApplication: com.tns.NativeScriptException: metadata file (treeStringsStream.dat) couldn't be opened! (Error: 2) 

StackTrace:
java.lang.RuntimeException: Unable to create application com.tns.NativeScriptApplication: com.tns.NativeScriptException: metadata file (treeStringsStream.dat) couldn't be opened! (Error: 2) 
	at android.app.ActivityThread.handleBindApplication(ActivityThread.java:6717)
	at android.app.ActivityThread.access$1300(ActivityThread.java:237)
	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1913)
	at android.os.Handler.dispatchMessage(Handler.java:106)
	at android.os.Looper.loop(Looper.java:223)
	at android.app.ActivityThread.main(ActivityThread.java:7656)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:592)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:947)
Caused by: com.tns.NativeScriptException: metadata file (treeStringsStream.dat) couldn't be opened! (Error: 2) 
	at com.tns.Runtime.initNativeScript(Native Method)
	at com.tns.Runtime.init(Runtime.java:642)
	at com.tns.Runtime.init(Runtime.java:616)
	at com.tns.Runtime.initRuntime(Runtime.java:606)
	at com.tns.Runtime.initializeRuntimeWithConfiguration(Runtime.java:581)
	at com.tns.RuntimeHelper.initRuntime(RuntimeHelper.java:164)
	at com.tns.NativeScriptApplication.onCreate(NativeScriptApplication.java:19)
	at android.app.Instrumentation.callApplicationOnCreate(Instrumentation.java:1192)
	at android.app.ActivityThread.handleBindApplication(ActivityThread.java:6712)
	... 8 more
```

farfromrefug, 2021-04-15:
@Mahtaran sorry but it does not seem to be related to this plugin.
you should first clean and rebuild. and if it still fails seek help on discod.

Mahtaran, 2021-04-15:
Hi! Thanks for your fast reply. I thought it would be related to this plugin since it only happened when I added this plugin, and disappeared as soon as I removed it. Tried cleaning and rebuilding a bunch of times, did not really have any effect, but I will try seeking help on Discord.

farfromrefug, 2024-03-11:
@mahtaran did you get this fixed ?

mahtaran, 2024-03-11:
> @mahtaran did you get this fixed ?

To be honest, no clue, I quit the job where I used NativeScript over two years ago. I remember never getting it to work quite as I wanted to though.

farfromrefug, 2024-03-12:
@mahtaran sorry to hear it didn't work out. 
thanks for closing issues though.
wish you the best